openbmc/openbmc #2296: The name xyz.openbmc_project.State.BMC was not provided by any .service files

gkeishin, 2017-09-13:
```
Aug 28 19:35:20 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.BMC was not provided by any .service files
Aug 28 19:35:21 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.Host was not provided by any .service files
Aug 28 19:35:22 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.Chassis was not provided by any .service files
Aug 28 19:35:30 witherspoon phosphor-dump-manager[1008]: Mon Aug 28 19:35:30 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/1
Aug 28 19:35:32 witherspoon phosphor-dump-manager[1008]: Mon Aug 28 19:35:32 UTC 2017 Sucessfully completed

/xyz/openbmc_project/dump/entry/enumerate
{
    "/xyz/openbmc_project/dump/entry/1": {
        "Elapsed": 1503948920,
        "Size": 4580
    }
}

root@witherspoon:~# ls -larth /var/lib/phosphor-debug-collector/dumps/1/obmcdump_1_1503948920.tar.xz
-rw-r--r--    1 root     root        4.5K Aug 28 19:35 /var/lib/phosphor-debug-collector/dumps/1/obmcdump_1_1503948920.tar.xz
root@witherspoon:~#
```

`Driver info`
```

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-168"
VERSION_ID="v1.99.9-168-gc180a26"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-168"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-b95f53ea200a6573035a25ab6ddebe917635eeb6 #1 Wed Sep 13 02:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 19:38:11 up 4 min,  load average: 1.13, 2.04, 1.01
root@witherspoon:~# cat /etc/timestamp
20170913022303
root@witherspoon:~# cat /media/pnor-ro-8764baa3/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.54
        op-build-v1.7-1321-g6eef393-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.8-rc1-pd9d2012
        hostboot-c68be97
        linux-4.12.8-openpower1-p919e663
        petitboot-v1.5.1-p848cad7
        machine-xml-c1e49a6-pbb9f626
        occ-854999a
        hostboot-binaries-67a15fd
        capp-ucode-9c73e9f
        sbe-4a62126-p180ac2a
root@witherspoon:~#
```

- [x] Journald 
[journald_elog_on_standby.txt](https://github.com/openbmc/openbmc/files/1298076/journald_elog_on_standby.txt)

- [x] BMC dump 
[obmcdump_1_1503948920.tar.zip](https://github.com/openbmc/openbmc/files/1298077/obmcdump_1_1503948920.tar.zip)

ojayanth, 2017-09-13:
These journal messages are expected if the dump is triggerd prior to required service starts. In this case state service started after dump is triggered.

Aug 28 19:35:18 witherspoon phosphor-fru-fault-monitor[1140]: /xyz/openbmc_project/logging/entry/1
Aug 28 19:35:20 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.BMC was not provided by any .service files
Aug 28 19:35:21 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.Host was not provided by any .service files
Aug 28 19:35:22 witherspoon phosphor-dump-manager[1008]: The name xyz.openbmc_project.State.Chassis was not provided by any .service files
Aug 28 19:35:23 witherspoon systemd[1]: Started Read OpenPOWER-format VPD from EEPROM.
Aug 28 19:35:23 witherspoon systemd[1]: Started Phosphor MAX31785 MSL.
Aug 28 19:35:30 witherspoon phosphor-dump-manager[1008]: Mon Aug 28 19:35:30 UTC 2017 Report is available in /var/lib/phosphor-debug-collector/dumps/1
Aug 28 19:35:32 witherspoon phosphor-dump-manager[1008]: Mon Aug 28 19:35:32 UTC 2017 Sucessfully completed
Aug 28 19:35:34 witherspoon systemd[1]: Started Wait for /org/openbmc/managers/System.
Aug 28 19:35:34 witherspoon systemd[1]: Starting Phosphor Reset Button0...
Aug 28 19:35:34 witherspoon systemd[1]: Starting Phosphor Power Button0...
Aug 28 19:35:34 witherspoon systemd[1]: Starting Phosphor BMC State Manager...
Aug 28 19:35:34 witherspoon systemd[1]: Starting Phosphor Chassis State Manager...
Aug 28 19:35:35 witherspoon systemd[1]: Starting Phosphor Power0 Control...

gkeishin, 2017-09-13:
So we need to fix this race condition to handle it better and not create dump for in @ojayanth

ojayanth, 2017-09-13:
This is not really race condition,  dreport collects the data at failing context and for any failure due to bad system state , should ignore.

gkeishin, 2017-09-13:
Like error log, if there is a dump generated its a concern for us.. so either don't log the dump if it's a no deal harm or fix the logic to sync it..

ojayanth, 2017-09-13:
Dump is created due to internalFailure reported by mboxd.
Refer Dump for details on the failure.
From Proc_Journal.og.
on 2017-08-28 19:35:08.097382 UTC [s=0482b1f029404728b7f6f78c67da7a39;i=2c2;b=6b4e9f75ff44485fa7d4770946350fe3;m=561f177;t=557d5646d78c8;x=e39dafc490c5622c]
    _BOOT_ID=6b4e9f75ff44485fa7d4770946350fe3
    _MACHINE_ID=d77bdfede38e4991bae5bd832d415646
    _HOSTNAME=witherspoon
    _UID=0
    _GID=0
    _SYSTEMD_SLICE=system.slice
    _TRANSPORT=journal
    _CAP_EFFECTIVE=3fffffffff
    PRIORITY=3
    CODE_FILE=/usr/include/phosphor-logging/log.hpp
    CODE_LINE=100
    CODE_FUNC=helper_log
    MESSAGE=The operation failed internally.
    TRANSACTION_ID=3394759793
    SYSLOG_IDENTIFIER=mboxd
    _PID=1164
    _COMM=mboxd
    _EXE=/usr/sbin/mboxd
    _CMDLINE=mboxd --flash 64M --window-size 1M
    _SYSTEMD_CGROUP=/system.slice/mboxd.service
    _SYSTEMD_UNIT=mboxd.service
    _SYSTEMD_INVOCATION_ID=3a0a701ca500417496b796b7000b123b
    _SOURCE_REALTIME_TIMESTAMP=1503948908097382

gkeishin, 2017-09-13:
Ok thanks.. but we will need to see about the dependencies created on state manager services. where dump is supposed to be independent and start early on boot as I understood from the discussion.

gkeishin, 2017-09-18:
alright.. as long this doesn't create further issue ahead. good to close..

closed 2017-09-18 by gkeishin