openbmc/openbmc #2297: Error log code location metadata is not useful

amboar, 2017-09-13:
Here's an example from the journal, stolen from #2295 but observed in a number of other cases:

```
{
        "__CURSOR" : "s=0482b1f029404728b7f6f78c67da7a39;i=2c2;b=6b4e9f75ff44485fa7d4770946350fe3;m=561f177;t=557d5646d78c8;x=e39dafc490c5622c",
        "__REALTIME_TIMESTAMP" : "1503948908099784",
        "__MONOTONIC_TIMESTAMP" : "90304887",
        "_BOOT_ID" : "6b4e9f75ff44485fa7d4770946350fe3",
        "_MACHINE_ID" : "d77bdfede38e4991bae5bd832d415646",
        "_HOSTNAME" : "witherspoon",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "PRIORITY" : "3",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "MESSAGE" : "The operation failed internally.",
        "TRANSACTION_ID" : "3394759793",
        "SYSLOG_IDENTIFIER" : "mboxd",
        "_PID" : "1164",
        "_COMM" : "mboxd",
        "_EXE" : "/usr/sbin/mboxd",
        "_CMDLINE" : "mboxd --flash 64M --window-size 1M",
        "_SYSTEMD_CGROUP" : "/system.slice/mboxd.service",
        "_SYSTEMD_UNIT" : "mboxd.service",
        "_SYSTEMD_INVOCATION_ID" : "3a0a701ca500417496b796b7000b123b",
        "_SOURCE_REALTIME_TIMESTAMP" : "1503948908097382"
```
Specifically, the metadata for the code location point to `helper_log()`, not to the location of the invocation of `helper_log()`

vishwabmc, 2017-10-26:
I put this in the channel
```
I am working on a bug that says the CODE_FUNC in log meta always says "helper_log".
Debugging, saw that is because journal_send is done from that function in log.hpp
So the need now is to know "who called log<>"

Obvious but a large change is to add CODE_FUNC in all code that invoke log<> and that does not seem recommended IMO.

I did good amount of search and the solutions I compiled are :

`https://stackoverflow.com/questions/353180/how-do-i-find-the-name-of-the-calling-function` has all good suggestions.

1/. User glibc runtime backtrace() utilities --> Adds a runtime overhead and not recommended
2/. Modify the function that has __builtin_FUNCTION()
Ex: void foo(const char* str= __builtin_FUNCTION()) --> this is working fine on my test setup but fails to compile on logging with mismatched argument. Need to debug more.. does not look like obvious ones.

3/. Doing undef of the original API and then redefining it ( Solution in that stackoverflow link with 32 votes ).

--> That works ( the operator() approach ) fine as long it is non templated but I am having trouble getting that version to  work on templated versions.

Basically, the part :
#undef FunctionName
#define FunctionName(...) \
    Reporter(__FUNCTION__, __LINE__)

-OR- ( not making sense ).
#undef FunctionName
#define FunctionName(...) \
    template <typename ...T> \
    Reporter(__FUNCTION__, __LINE__)

Well, this compiles, but then clearly not invoking the Reporter ;)

Since this is a pre-processor substitution, need a way to pass the template args as in Reporter<...>(__FUNCTION__, __LINE__, )(args...)

Any thoughts on these in terms of 

1/. Which approach seems better ? ( the second one seems small foot print but then we may have an issue if we want CODE_LINE
2/. Any solutions to the template issue ?
```

geissonator, 2017-12-14:
Lots of good investigation info from Vishwa above, we'll need to revisit this one in the future.  Moving out of current sprint.

amboar, 2019-03-27:
While stale, no-one has actually fixed the issue - the information in the logs is still useless.

spinler, 2019-07-19:
For reasons outside of my control, I had to open an issue in another spot to get eyes on this issue: https://github.com/ibm-openbmc/dev/issues/1039.

spinler, 2020-07-20:
Not giving up on this one yet.

geissonator, 2020-07-20:
I know Vernon had mentioned https://en.cppreference.com/w/cpp/experimental/source_location as a possibility in c++20

closed 2021-06-09 by stale[bot]