openbmc/openbmc #2298: IPMID : Invalid event data  on Poweron

gkeishin, 2017-09-13:
`Journald snippet`
```
Aug 28 20:06:00 witherspoon ipmid[1208]: Invalid event data
```

```

/xyz/openbmc_project/state/host0
{
    "AttemptsLeft": 3,
    "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
    "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Reboot"
}
```

`Driver Info`
```

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-168"
VERSION_ID="v1.99.9-168-gc180a26"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-168"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-b95f53ea200a6573035a25ab6ddebe917635eeb6 #1 Wed Sep 13 02:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 19:38:11 up 4 min,  load average: 1.13, 2.04, 1.01
root@witherspoon:~# cat /etc/timestamp
20170913022303
root@witherspoon:~# cat /media/pnor-ro-8764baa3/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.54
        op-build-v1.7-1321-g6eef393-dirty
        buildroot-2017.05-163-g8035745
        skiboot-v5.8-rc1-pd9d2012
        hostboot-c68be97
        linux-4.12.8-openpower1-p919e663
        petitboot-v1.5.1-p848cad7
        machine-xml-c1e49a6-pbb9f626
        occ-854999a
        hostboot-binaries-67a15fd
        capp-ucode-9c73e9f
        sbe-4a62126-p180ac2a
root@witherspoon:~#
```

- [x] Journald 
[journald_ipmid_error.txt](https://github.com/openbmc/openbmc/files/1298169/journald_ipmid_error.txt)

gkeishin, 2017-09-13:
@tomjoseph83 ^^^

tomjoseph83, 2017-10-02:
For the fw_boot_sensor, the values supported for eventData2 as per Witherspoon MRW are `0x01,0x03,0x13,0x14` , but hostboot is sending 0x07 as well in the eventData2, which results in the Invalid event data trace.

gkeishin, 2017-10-16:
Fix coming up for this one ? @tomjoseph83

rfrandse, 2017-10-25:
<https://gerrit.openbmc-project.xyz/7539> Add missing boot progress sensor value
Resolves  #2298 IPMID : Invalid event data  on Poweron

closed 2017-10-26 by zahrens