openbmc/openbmc #2300: BMC Inventory boxelder Serial Number Empty

gkeishin, 2017-09-13:
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc": {
        "BuildDate": "",
        "FieldReplaceable": 0,
        "Manufacturer": "IBM",
        "Model": "",
        "PartNumber": "01DH118",
        "Present": 1,
        "PrettyName": "BMC PLANAR  ",
        "SerialNumber": "000000000000",
        "UUID": ""
    },
```

I see all other CPU/DIMM/host Planner objects are populated with Serial number.. So don't we need this one too updated ?

geissonator, 2017-09-13:
Did you verify it was written in VPD?  Probably have to use cronus to do it.

gkeishin, 2017-09-18:
I ll talk to MFG guys on how to set this one..  and verify

gkeishin, 2017-09-22:
We did the trial with the help of local MFG team. 

We updated the SN keyword in the backplane VPD with 0xdead

```
00000090  80 00 52 54 04 56 49 4e  49 44 52 0c 42 4d 43 20  |..RT.VINIDR.BMC |
000000a0  50 4c 41 4e 41 52 20 20  46 4e 07 30 30 30 30 30  |PLANAR  FN.00000|
000000b0  30 30 50 4e 07 30 31 44  48 31 31 38 53 4e de ad  |00PN.01DH118SN..|
```
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~#

/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc
{
    "BuildDate": "",
    "FieldReplaceable": 0,
    "Manufacturer": "IBM",
    "Model": "",
    "PartNumber": "01DH118",
    "Present": 1,
    "PrettyName": "BMC PLANAR  ",
    "SerialNumber": "000000000000",
    "UUID": ""
}
```
We power on and boot the host.. but still not read and not updated in inventory.

gkeishin, 2017-09-22:
@tomjoseph83 ^^^

devenrao, 2017-10-24:
I had logged the value read from VPD for Serial number an it is showing as "000000000000" so it is updating properly.
Probably VPD need to be fixed with valid serial number

```
Oct 24 06:50:01 witherspoon openpower-read-vpd[1654]: SerialNumber000000000000
Oct 24 06:50:01 witherspoon openpower-read-vpd[1654]: PartNumber01DH137
Oct 24 06:50:01 witherspoon openpower-read-vpd[1654]: ManufacturerIBM
Oct 24 06:50:01 witherspoon openpower-read-vpd[1654]: PrettyNameBMC PLANAR
Oct 24 06:50:01 witherspoon openpower-read-vpd[1654]: MACAddress70:e2:84:14:33:65
```
```
        xyzAssetProps["SerialNumber"] =
            vpdStore.get<Record::VINI, record::Keyword::SN>();
        std::cout << "SerialNumber" << vpdStore.get<Record::VINI, record::Keyword::SN>() << std::endl;
```

Assume noting else needs to be done here

gkeishin, 2017-10-24:
@devenrao If u can write some data to it then, I can verify and close this one

devenrao, 2017-10-25:
Rajesh says there is no direct way to update the SN, last time he tweaked the MAC address code to update the SN in the bmc card.

dkodihal, 2017-10-26:
I updated the vpd manually and the SN shows up fine.

```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc": {
      "BuildDate": "", 
      "FieldReplaceable": 0, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01DH137", 
      "Present": 1, 
      "PrettyName": "BMC PLANAR  ", 
      "SerialNumber": "010101010101", 
      "UUID": ""
    }
```

closed 2017-10-26 by dkodihal

gkeishin, 2017-10-26:
Perfect.. Thanks, . Can u help update on my system too @dkodihal  :) or perhaps share steps to do so on our other boxes