openbmc/openbmc #2302: BMC dump core dump not captured

gkeishin, 2017-09-13:
@ojayanth and I tested this one on Master and it doesnt work.. but he is trying to recreate to understand the root cause

```
Aug 28 19:54:08 witherspoon systemd[1]: Created slice system-systemd\x2dcoredump.slice.
Aug 28 19:54:08 witherspoon systemd[1]: Started Process Core Dump (PID 1340/UID 0).
Aug 28 19:54:08 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
Aug 28 19:54:08 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Aug 28 19:54:08 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Aug 28 19:54:08 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
Aug 28 19:54:08 witherspoon systemd[1]: Stopped Phosphor Inband IPMI.
Aug 28 19:54:08 witherspoon systemd[1]: Started Phosphor Inband IPMI.
Aug 28 19:54:19 witherspoon systemd-coredump[1341]: Process 1195 (ipmid) of user 0 dumped core.
```

`Dump not generated`

```

/xyz/openbmc_project/dump/enumerate
{
    "/xyz/openbmc_project/dump/internal/manager": {}
}

root@witherspoon:~# df /var/lib/systemd/coredump
Filesystem           1K-blocks      Used Available Use% Mounted on
tmpfs                   216120       104    216016   0% /var/lib/systemd/coredump
root@witherspoon:~#
```

`Driver Info`

````
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-168"
VERSION_ID="v1.99.9-168-gc180a26"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-168"
BUILD_ID="v1.99.9"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-b95f53ea200a6573035a25ab6ddebe917635eeb6 #1 Wed Sep 13 02:20:54 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# uptime
 19:38:11 up 4 min,  load average: 1.13, 2.04, 1.01
root@witherspoon:~# cat /etc/timestamp
20170913022303
root@witherspoon:~#
```

gkeishin, 2017-09-13:
Probably similar to https://github.com/openbmc/openbmc/issues/2287

ojayanth, 2017-09-14:
Issue is same as #2287

gkeishin, 2017-09-14:
@dkodihal and we also had another opinion on it during the scrum and finding other solution in the mean time using this

ojayanth, 2017-09-16:
https://github.com/openbmc/openbmc/issues/2240 is used for adding temporary workaround . and actual issue is tracked under. https://github.com/openbmc/openbmc/issues/2287

geissonator, 2017-09-18:
Core dumps disabled for now, moving to 5.1

ojayanth, 2017-09-18:
@gkeishin agreed to close this issue.

closed 2017-09-18 by gkeishin