openbmc/openbmc #2305: Fan Control: Use tach feedback instead of target on speed changes

msbarth, 2017-09-13:
To allow smaller speed change delta's, the addition of a defined action to update the current increase/decrease requested speed to the lowest value of a group of sensor properties(in this case, a group of tach feedback properties).

msbarth, 2017-09-14:
- [ ] Create a public function that set/update the base requested target speed
- [ ] Create an action function that will update the base requested target speed from the lowest value from a given group of sensors
- [ ] Add an event that contains the valid tach feedbacks as the group with the action created updating the base requested target speed to the lowest feedback value

rfrandse, 2017-09-18:
<https://gerrit.openbmc-project.xyz/6706> witherspoon - Request speed based on tach feedback
Resolves  #2305 Fan Control: Use tach feedback instead of target on speed changes

closed 2017-10-02 by geissonator