openbmc/openbmc #2308: CPU and DIMM info is not in inventory on zaius

ryandcyu, 2017-09-14:
@anoo1 

I see the /xyz/openbmc_project/inventory, It only has cpu0 and cpu1 information
```   
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0",
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1",
    "/xyz/openbmc_project/inventory/system"
```
But /org/openbmc/inventory have so many item as below
```
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core18",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core19",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm28",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm29",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core10",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm23",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm20",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core13",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core14",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm27",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm24",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm25",
    "/org/openbmc/inventory/system/systemevent",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core22",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core23",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core20",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core21",
    "/org/openbmc/inventory/system/misc",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core18",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm31",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm30",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core19",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm8",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core13",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core12",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core11",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core10",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core17",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core16",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core15",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core14",
    "/org/openbmc/inventory/system/chassis/motherboard/bmc",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm22",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core9",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core8",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core7",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core6",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core5",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core4",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core3",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core2",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core1",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0/core0",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm21",
    "/org/openbmc/inventory/system/chassis/motherboard/refclock",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm26",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core11",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core15",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core16",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core17",
    "/org/openbmc/inventory/system/chassis/motherboard/todclock",
    "/org/openbmc/inventory/system/chassis",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core8",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core9",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core2",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core3",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core0",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core1",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core6",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core7",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core4",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core5",
    "/org/openbmc/inventory/system/chassis/motherboard/pcieclock",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core21",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core20",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core23",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core22",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu0",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm9",
    "/org/openbmc/inventory/system",
    "/org/openbmc/inventory/system/bios",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm3",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm2",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm1",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm0",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm7",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm6",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm5",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm4",
    "/org/openbmc/inventory/system/chassis/fan4",
    "/org/openbmc/inventory/system/chassis/fan5",
    "/org/openbmc/inventory/system/chassis/fan2",
    "/org/openbmc/inventory/system/chassis/fan3",
    "/org/openbmc/inventory/system/chassis/fan0",
    "/org/openbmc/inventory/system/chassis/fan1",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm17",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm16",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm15",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm14",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm13",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm12",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm11",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm10",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm19",
    "/org/openbmc/inventory/system/chassis/motherboard/dimm18",
    "/org/openbmc/inventory/system/chassis/motherboard/cpu1/core12",
    "/org/openbmc/inventory/system/chassis/motherboard"
```

```
The BMC version as below
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-52"
VERSION_ID="v1.99.9-52-g06faefdb-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-52"
BUILD_ID="v1.99.9"
```
I am aware of /org/openbmc would be deprecated, So these all items will move to /xyz/openbmc_project/inventory? When will completed on zaius?

geissonator, 2017-09-14:
I did the initial support to get the xyz inventory into Zaius with https://gerrit.openbmc-project.xyz/#/c/5338/   This would need to be built upon to add in support for the rest of the parts.

ryandcyu, 2017-10-23:
I add in by myself, close it.

closed 2017-10-23 by ryandcyu