openbmc/openbmc #2309: Get pnor information via restful api

ryandcyu, 2017-09-14:
@anoo1 

Currently, we can get the BMC information via restful api
```
/xyz/openbmc_projet/software/aaf838cc
{
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
    "Path": "",
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
    "Version": "v1.99.9-52-g06faefdb-dirty"
  },
```
Have any plan to implement the PNOR information?

anoo1, 2017-09-26:
@agangidi53 will follow-up to see who could implement a simple pnor version manager that would create the software dbus object with the version of a pnor chip in ffs format.

KennethWilke, 2017-09-26:
There is a way to grab the pnor through pflash:
```
root@zaius:~# pflash -i
Flash info:
-----------
Name          = /dev/mtd6
Total size    = 64MB 
Erase granule = 64KB 

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000) 
ID=01             HBB 00010000..000a0000 (actual=00090000) [ECC]
ID=02            HBEL 000a0000..000c4000 (actual=00024000) [ECC]
ID=03           GUARD 000d0000..000d5000 (actual=00005000) [ECC]
ID=04             HBD 000e0000..00200000 (actual=00120000) [ECC]
ID=05           DJVPD 00200000..00248000 (actual=00048000) [ECC]
ID=06            MVPD 00250000..002e0000 (actual=00090000) [ECC]
ID=07            CVPD 002e0000..00328000 (actual=00048000) [ECC]
ID=08             HBI 00330000..00f90000 (actual=00c60000) [ECC]
ID=09             SBE 00f90000..01012000 (actual=00082000) [ECC]
ID=10           HCODE 01020000..01140000 (actual=00120000) [ECC]
ID=11            HBRT 01140000..015c0000 (actual=00480000) [ECC]
ID=12         PAYLOAD 015c0000..016c0000 (actual=00100000) 
ID=13      BOOTKERNEL 016c0000..025c0000 (actual=00f00000) 
ID=14           NVRAM 025c0000..02650000 (actual=00090000) 
ID=15             OCC 02650000..02770000 (actual=00120000) [ECC]
ID=16         FIRDATA 02770000..02773000 (actual=00003000) [ECC]
ID=17            CAPP 02780000..027a4000 (actual=00024000) [ECC]
ID=18         SECBOOT 027b0000..027d4000 (actual=00024000) [ECC]
ID=19         BMC_INV 027e0000..027e9000 (actual=00009000) 
ID=20            HBBL 027f0000..027f6000 (actual=00006000) [ECC]
ID=21        ATTR_TMP 02800000..02808000 (actual=00008000) 
ID=22       ATTR_PERM 02810000..02818000 (actual=00008000) [ECC]
ID=23         VERSION 02820000..02821000 (actual=00001000) 
ID=24     IMA_CATALOG 02830000..02870000 (actual=00040000) [ECC]
ID=25         RINGOVD 02870000..02890000 (actual=00020000) 
ID=26         WOFDATA 02890000..02b90000 (actual=00300000) [ECC]
ID=27     HB_VOLATILE 02b90000..02b95000 (actual=00005000) [ECC]
ID=28     BACKUP_PART 03ff7000..03fff000 (actual=00000000) 
root@zaius:~# pflash -P VERSION -r /tmp/pnor-version
Reading to "/tmp/pnor-version" from 0x02820000..0x02821000 !
[==================================================] 100%
root@zaius:~# cat /tmp/pnor-version 
OpenPower-BarreleyeG2-v0.1.0
	op-build-v1.17-195-g4dec09a
	buildroot-2017.05-93-g9b91216
	skiboot-v5.7-rc2
	hostboot-986dd16
	linux-4.12.1-openpower1-pb6b658a
	petitboot-v1.5.0-p6906565
	machine-xml-a3dc067
	occ-2f3397d
	hostboot-binaries-c24fccd
	capp-ucode-9c73e9f
	sbe-02021c6-p7968a86
```

I'm not sure what it would take to add a dbus object that could pull this data

anoo1, 2017-10-03:
hi @KennethWilke , a few pointers:
The repo that creates the pnor dbus objects is: https://github.com/openbmc/openpower-pnor-code-mgmt
This package is only added to systems that have virtual pnor enabled, so there could be a small binary that could be added to all other systems.

As a reference, the bmc dbus objects are created by: https://github.com/openbmc/phosphor-bmc-code-mgmt
we're adding support for creating dbus-objects to non-ubi systems with: https://gerrit.openbmc-project.xyz/#/c/6734/
which can be used as a guideline for the pnor change.

mine260309, 2018-08-15:
@gtmills It only works on Witherspoon that enabled vpnor.
For other machines, it is still not supported.

closed 2019-04-26 by stale[bot]