openbmc/openbmc #2310: Sprint 17.09.20 - Committed

rfrandse, 2017-09-14:


dkodihal, 2017-09-15:
***Squad 1 Sprint Plan***

spinler, 2017-09-18:
***Squad 2 Sprint Plan***

**Matt S**

**Matt B**

**Gunnar**

**Chris**

**Brandon**

**Eddie**

anoo1, 2017-09-24:
***Squad 3 Sprint Plan***

**Adriana**
#2028 Shield setting inventory property attributes by external interfaces

**Saqib**
#1944 BMC: Block BMC reboot during u-boot update
#1896 Performance Optimization to query Software Version Redundancy Priority

**Michael**
#2175 Mbox service core-dump on BMC r/r during powering on
#2264 Software Object: Provide method to delete all versions
#2249 BMC new code Activation: Error: "rwreset" not defined

**Chris**

**Leo**
#1663 Rename mboxd dbus object to be in correct xyz.openbmc_project
#1539 Update rest_dbus.py MethodHandler Class to Convert Method Arguments

**Charlie**

closed 2017-10-12 by rfrandse