openbmc/openbmc #2320: Persistency with Cereal : implementing versioning

dkodihal, 2017-09-15:
Cereal has versioning support. This is required when via a code update, we're changing the structures of objects we persist.

This story is to investigate what support Cereal offers, and then implement versioning on the error log object. Bonus would be to do the same for inventory and settings objects.

sivassrr, 2017-09-25:
Evaluated this from FVT perspective and got confirmation from @vishwabmc that it wouldnt impact to FVT. We can run regression associate with code update.

vishwabmc, 2017-09-25:
@sivassrr When we really split the errorlog properties into multiple interfaces ( work that @ratagupt is doing ), I should be able to give you a better picture if FVT is testing the properties of the interface

vishwabmc, 2017-09-25:
# **TASK**

vishwabmc, 2017-09-25:
Currently, all the properties are under `xyz.openbmc_project.Logging.Entry` but the Eventing may twist this a bit and will see @ratagupt

vishwabmc, 2017-09-25:
Perhaps, the initial code could be to just update the elog_serialize to contain the versioning

vishwabmc, 2017-09-26:
# **TASKS**

- [ ] Make changes to elog_serialize to accept the version parameter
- [ ] Update class versioning in configure.ac and use it in the elog_entry
- [ ] Add code to remove the entry file on failure to de-deserialize
- [ ] When @ratagupt splits the properties into interfaces, make changes the way it's stored.

@dkodihal Do you see any other thing ?

rfrandse, 2017-09-29:
<https://gerrit.openbmc-project.xyz/7009> Enable Cereal class versioning
Fixes  #2320 Persistency with Cereal : implementing versioning

dkodihal, 2017-10-03:
@vishwabmc I looked at the code you've uploaded. Mind explaining how a typical version bump scenario would work? For eg now I see you register classes implementing the Cereal APIs with a version =1. Let's say there's an interface change, what are we expected to change then? Just bump the version in configure.ac and Cereal will take care of a potential mismatch in versions of persisted objects and new objects (by new I mean post a non-disruptive code udpate)?

vishwabmc, 2017-10-04:
@dkodihal Cereal gives a mechanism that enables application writers to take decision based on current version of the class vs version of the serialized object. So application writers need to handle the differences,.

As part of `save(class Archive, const std::uint32_t version)`, the `$version` gives the `version that is currently being serialized`

As part of `load(class Archive, const std::uint32_t version)`, the `$version` gives the `version that is previously serialized`.

So inside `load`, we can say `if $version < condition>`, then ........

closed 2017-10-11 by geissonator