openbmc/openbmc #2325: occ hwmon is disappear after BMC reboot.

ryandcyu, 2017-09-18:
@anoo1 
After host power on, BMC has homon node for occ
ex. in my system(zaius), it will show under /sys/class/hwmon/hwmon4
```
root@zaius:~# ls -l /sys/class/hwmon/
total 0
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon0 -> ../../devices/platform/ahb/ahb:apb/1e786000.pwm-tacho-controller/hwmon/hwmon0
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon1 -> ../../devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a300.i2c-bus/i2c-7/i2c-215/215-0054/hwmon/hwmon1
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon2 -> ../../devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a040.i2c-bus/i2c-0/0-0064/hwmon/hwmon2
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon3 -> ../../devices/platform/iio-hwmon/hwmon/hwmon3
lrwxrwxrwx    1 root     root             0 Sep  4 09:08 hwmon4 -> ../../devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:06/sbefifo1-dev0/occ1-dev0/hwmon/hwmon4
```

But after BMC reboot (Host was still power on), the hwmon of occ would be disappear
```
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon0 -> ../../devices/platform/ahb/ahb:apb/1e786000.pwm-tacho-controller/hwmon/hwmon0
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon2 -> ../../devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a300.i2c-bus/i2c-7/i2c-215/215-0054/hwmon/hwmon2
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon3 -> ../../devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a040.i2c-bus/i2c-0/0-0064/hwmon/hwmon3
lrwxrwxrwx    1 root     root             0 Sep  4 09:05 hwmon4 -> ../../devices/platform/iio-hwmon/hwmon/hwmon4
```
I fetched some log about OCC
```
Sep 04 09:06:24 zaius systemd[1]: Starting OpenPOWER OCC controller...
Sep 04 09:06:29 zaius phosphor-network-manager[1072]: Exception occured during getting of DHCP value
Sep 04 09:06:36 zaius systemd[1]: Started OpenPOWER OCC controller.
Sep 04 09:07:00 zaius systemd[1]: Created slice system-op\x2docc\x2denable.slice.
Sep 04 09:07:00 zaius systemd[1]: Starting OpenPOWER OCC Active Enable for host 0...
Sep 04 09:07:02 zaius systemd[1]: Started OpenPOWER OCC Active Enable for host 0.
Sep 04 09:07:08 zaius kernel: occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Sep 04 09:07:08 zaius kernel[1030]: [  116.920000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Service hold-off time over, scheduling restart.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Stop job pending for unit, delaying automatic restart.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Service hold-off time over, scheduling restart.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Stop job pending for unit, delaying automatic restart.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Service hold-off time over, scheduling restart.
Sep 04 09:07:11 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Stop job pending for unit, delaying automatic restart.
```
It seems that hwmon of occ was stop while BMC reboot, but didn't start again while BMC come back.

Tested BMC version
```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-4"
VERSION_ID="v1.99.10-4-ge146f85"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-4"
BUILD_ID="v1.99.10"
```

anoo1, 2017-09-19:
@geissonator mentions this has been fixed in witherspoon but a fix for zaius might need to be ported.

anoo1, 2017-09-19:
The commit was: https://gerrit.openbmc-project.xyz/#/c/6169/
which is common code, so there may be something in zaius missing to enable the occ on reboot.

ryandcyu, 2017-09-21:
@GHF @anoo1 I have verified on v1.99.10, After resetting the BMC, I saw the kernel message "failed to get OCC poll response: -19" and then hwmon of occ was gone, but when I bind occ driver again, the hwmon came back and occ function was workable.
I saw the something message relate to occ during BMC reset as below.
```
Sep 04 09:06:51 zaius phosphor-bmc-state-manager[1101]: BMC_READY
Sep 04 09:06:51 zaius phosphor-bmc-state-manager[1101]: Setting the BMCState field
Sep 04 09:06:51 zaius flash_bios.exe[1085]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Sep 04 09:06:51 zaius flash_bios.exe[1085]: flasher: , BMC = 1, address = 0x0
Sep 04 09:06:51 zaius flash_bios.exe[1085]: Setting up flash
[   96.780000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Sep 04 09:06:52 zaius systemd-timesyncd[964]: Timed out waiting for reply from 216.239.35.12:123 (time4.google.com).
Sep 04 09:06:52 zaius kernel: occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Sep 04 09:06:52 zaius kernel[1006]: [   96.780000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
Sep 04 09:06:53 zaius phosphor-ledmanager[1012]: Asserting LEDs
Sep 04 09:06:53 zaius phosphor-ledmanager[1012]: No service providers for physical LED
Sep 04 09:06:53 zaius flash_bios.exe[1085]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Sep 04 09:06:53 zaius flash_bios.exe[1085]: flasher: , BMC = 1, address = 0x0
Sep 04 09:06:53 zaius flash_bios.exe[1085]: Setting up flash
Sep 04 09:06:54 zaius systemd[1]: Started Scan FSI devices.
Sep 04 09:06:54 zaius systemd[1]: Started Assert bmc_booted LED.
Sep 04 09:06:54 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Unit not needed anymore. Stopping.
Sep 04 09:06:54 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Service hold-off time over, scheduling restart.
Sep 04 09:06:54 zaius systemd[1]: xyz.openbmc_project.Hwmon@gpio\x2dfsi-cfam\x400\x2c0-sbefifo\x402400-occ\x401-occ\x2dhwmon\x401.service: Stop job pending for unit, delaying automatic restart.
Sep 04 09:06:54 zaius systemd[1]: Stopped Phosphor Hwmon Poller.
```

mine260309, 2017-09-22:
@vishwabmc Could you help to check this issue?
In short, it's on Zaius system, occ hwmon works fine on first boot, and after bmc reboot (host keeps running), occ hwmon disappears.

vishwabmc, 2017-09-22:
Sure. Will look into it.

vishwabmc, 2017-09-22:
@mine260309 could you check if occ-enable.sh is present in lib/systemd/system ?.  Btw are you saying it's not bound ?  That would be a different thing. Could you clarify which path is missing

ryandcyu, 2017-09-22:
@vishwabmc I did not see occ-enable.sh in lib/systemd/system. I'm not sure occ is bound or not, how can I check occ bound?

vishwabmc, 2017-10-18:
OCC-HWMON path is at : `/sys/bus/platform/drivers/occ-hwmon/`

For Master OCC, there must be a soft link `occ1-dev0` -> `/sys/bus/platform/devices/occ1-dev0` if bound.

Please update what do you see there. Also, do you think it's more of a documentation problem and a README would help etc ?

Also, do you see these ?
```
root@w:/lib/systemd/system# ls -l op-occ-enable\@.service 
-rw-r--r--    1 root     root           453 Oct 16 13:31 op-occ-enable@.service
root@w:/lib/systemd/system#

root@w:/lib/systemd/system# ls -l op-occ-disable\@.service 
-rw-r--r--    1 root     root           264 Oct 16 13:31 op-occ-disable@.service
root@w:/lib/systemd/system# 

root@w:/usr# ls -l /usr/bin/occ-active.sh 
-rwxr-xr-x    1 root     root           553 Oct 16 13:31 /usr/bin/occ-active.sh
root@w:/usr# 
```

a-barrera, 2017-10-18:
@vishwabmc I recreated this on my zaius system. Here's what I'm seeing:

```
root@zaius:/lib/systemd/system# ls -l /sys/bus/platform/drivers/occ-hwmon/
total 0
--w-------    1 root     root          4096 Oct  6 01:40 bind
lrwxrwxrwx    1 root     root             0 Oct  6 01:41 occ1-dev0 -> ../../../../devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:06/sbefifo1-dev0/occ1-dev0
--w-------    1 root     root          4096 Oct  6 01:38 uevent
--w-------    1 root     root          4096 Oct  6 01:41 unbind
```

and 

```
root@zaius:/lib/systemd/system# ls -l op-occ-enable\@.service
-rw-r--r--    1 root     root           543 Oct  6  2017 op-occ-enable@.service
root@zaius:/lib/systemd/system# ls -l op-occ-disable\@.service
-rw-r--r--    1 root     root           264 Oct  6  2017 op-occ-disable@.service
root@zaius:/lib/systemd/system# ls -l /usr/bin/occ-active.sh 
-rwxr-xr-x    1 root     root           553 Oct  6  2017 /usr/bin/occ-active.sh
```

eddiejames, 2017-10-18:
This is happening because the SBEFIFO device is failing to probe due to existing data in the FIFO. Since there's no SBEFIFO device, no OCC device can be created either, resulting in a failure to open the file.

There is a fix already being integrated that should resolve this by resetting the SBEFIFO if it sees data in there during probe.

GHF, 2017-10-18:
I just got around to reading this issue. Is the SBEFIFO bug specific to Zaius, then?

eddiejames, 2017-10-18:
The SBEFIFO bug was not specific to Zaius. But it looks like there is another issue here, perhaps with occ controller service, as Adrian is still seeing the problem with the SBEFIFO fix.

geissonator, 2017-10-18:
So we patched on Eddie's fix for the SBEFIFO thing, and now we're at consistently seeing hwmon show up for only the second processor, but not the first.  One theory is the initial query done by occ-control to see if it is already bound is returning true for first proc, but not seeing any smoking gun yet.

eddiejames, 2017-10-25:
Got a fix up for this one. Need to make sure the openpower-occ-control app is in sync after fsi rescans.

rfrandse, 2017-10-26:
<https://gerrit.openbmc-project.xyz/7587> Add additional device bind case if occActive is out-of-sync
Resolves  #2325 occ hwmon is disappear after BMC reboot.

closed 2017-11-13 by geissonator

rlippert, 2017-12-04:
Can someone provide a link for "Eddie's fix for the SBEFIFO thing"?

I'd like to backport to our internal tree and verify the fix.  Thanks!

eddiejames, 2017-12-04:
Here is the commit into openpower-occ-control: 
https://github.com/openbmc/openpower-occ-control/commit/5e177970328acb51d7213d137b6b47cde0451780

Here is the bump of the openbmc recipe:
https://github.com/openbmc/openbmc/commit/d93ae5f71957c103e22167263d7bfec1f7b44a69