openbmc/openbmc #2327: phosphor-hwmon: remove sensors that fail with EAGAIN from bus

spinler, 2017-09-18:
In the design call today, it was decided to treat sensors that fail with EAGAIN differently than other failures.

If a sensor read fails with EAGAIN past the retry threshold, it should be removed from D-Bus.  Any clients that are monitoring that sensor can register for the InterfacesRemoved signal on it to know when it goes away and act accordingly.  No hardware callout will be made.

msbarth, 2018-03-20:
A couple ideas to consider in supporting this would be:

1. Add an additional command line parameter to readd called "Remove on RC" that would take an optional return code and a list of sensors where when that return code is received on any of the sensors listed, the dbus object would be removed. (This parameter would have the rc and list of sensors optional so it could support replacing the current configure flag used to remove any/all sensors from dbus to a per hwmon instance basis).
```
  phosphor-hwmon-readd --rm_on_rc [RC [SENSORS ...]]
```

2. For each sensor within the hwmon device conf file, a "Remove on RC" entry could be set to a particular return code where that sensor would be removed from dbus when that return code is returned.
```
  LABEL_temp198 = "gpu0_core_temp"
  REMOVERC_temp198 = "EAGAIN"
```

msbarth, 2018-03-21:
It seems from the service file setup that adding an additional command line parameter is not feasible with how each hwmon instance is started. Going to propose #2 as the best solution to the mailing list.

msbarth, 2018-03-23:
- [ ] Define sensor removal return codes within hwmon device conf file
- [ ] Store defined sensor return codes for dbus removal
- [ ] Remove sensors when given return codes are received
- [ ] Re-add sensors when return code is not longer present (keep attempting to read sensor)

rfrandse, 2018-03-29:
<https://gerrit.openbmc-project.xyz/9825> Re-add removed sensors during monitoring
Resolves  #2327 phosphor-hwmon: remove sensors that fail with EAGAIN from bus

closed 2018-04-13 by geissonator