openbmc/openbmc #2329: phosphor-hwmon: Set a Fault/Functional property when the sysfs fault property is set

spinler, 2017-09-18:
The hwmon documentation states:
```
Each input channel may have an associated fault file. This can be used
to notify open diodes, unconnected fans etc. where the hardware
supports it. When this boolean has value 1, the measurement for that
channel should not be trusted.
```

phosphor-hwmon should check for that fault status, if it exists, when reading a sensor and set a new D-Bus property on that sensor  (Maybe using the OperationalStatus interface) to reflect that fault.  The sensor object would not be removed from D-Bus, and the process should no longer exit.

Things to consider:
* keep retrying and checking for the fault to go away?
* even bother reading the sensor value if the fault is set?
* not all device drivers behind sensors may support the _fault file.  Does that mean the fault property should be set in D-Bus when the actual sensor value read fails?

msbarth, 2018-04-26:
When an associated fault file for a sensor exists, the OperationalStatus interface will be added with the Functional property set to the corresponding value within that fault file. The interface will not be added to sensors that do not provide an associated fault file.

If a sensors if marked faulted (non-functional), its associated input file will not be read. When the sensor is initially faulted, the value will default to 0 and at anytime during the monitoring loop a sensor is faulted, its value will not be read from the input file and updated on dbus.

- [ ] Add status interface to sensors
- [ ] Fault status check before reading sensor value
- [ ] Skip updating value for faulted sensors
- [ ] The fault will be checked/updated on each iteration of the monitoring loop

rfrandse, 2018-04-26:
<https://gerrit.openbmc-project.xyz/10315> Refresh sensor functional state
Resolves:  #2329 phosphor-hwmon: Set a Fault/Functional property when the sysfs fault property is set

closed 2018-05-02 by geissonator