openbmc/openbmc #2330: phosphor-fru-fault-monitor: Failed to get response from the method.

gkeishin, 2017-09-18:
```
Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [
        "_PID=1132"
    ],
    "Id": 1,
    "Message": "xyz.openbmc_project.Led.Mapper.Error.ObjectNotFoundError",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1505492759564,
    "associations": []
}
```

`journald snippet`
```

Sep 15 16:25:51 witherspoon phosphor-fru-fault-monitor[1132]: Failed to get response from the method.
Sep 15 16:25:59 witherspoon phosphor-fru-fault-monitor[1132]: /xyz/openbmc_project/logging/entry/1

{
        "__CURSOR" : "s=a2b21504ef6b4f6bbf0f46f5c001fd14;i=2da;b=eea725af758c45aca359993a93b2f398;m=55f0573;t=5593cd926f72f;x=689b992282ee9fbf",
        "__REALTIME_TIMESTAMP" : "1505492759672623",
        "__MONOTONIC_TIMESTAMP" : "90113395",
        "_BOOT_ID" : "eea725af758c45aca359993a93b2f398",
        "_MACHINE_ID" : "d77bdfede38e4991bae5bd832d415646",
        "_HOSTNAME" : "witherspoon",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
        "PRIORITY" : "3",
        "CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
        "CODE_LINE" : "100",
        "CODE_FUNC" : "helper_log",
        "TRANSACTION_ID" : "959732300",
        "SYSLOG_IDENTIFIER" : "phosphor-fru-fault-monitor",
        "_PID" : "1132",
        "_COMM" : "phosphor-fru-fa",
        "_EXE" : "/usr/sbin/phosphor-fru-fault-monitor",
        "_CMDLINE" : "phosphor-fru-fault-monitor",
        "_SYSTEMD_CGROUP" : "/system.slice/obmc-fru-fault-monitor.service",
        "_SYSTEMD_UNIT" : "obmc-fru-fault-monitor.service",
        "_SYSTEMD_INVOCATION_ID" : "699ae97a67bf4a4c81aa7c47583faebf",
        "MESSAGE" : "/xyz/openbmc_project/logging/entry/1",
        "_SOURCE_REALTIME_TIMESTAMP" : "1505492759669497"
}
```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-16"
VERSION_ID="v1.99.10-16-ga0ff17f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-16"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1 Mon Sep 18 15:30:45 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald  
[journald_standby_master.txt](https://github.com/openbmc/openbmc/files/1311838/journald_standby_master.txt)

dhruvibm, 2017-09-19:
Incorrectly logging an error when there is no error logs present, removing that error and pushing a patch.

rfrandse, 2017-09-19:
<https://gerrit.openbmc-project.xyz/6739> Remove incorrect error log when no errors present.
Resolves  #2330 phosphor-fru-fault-monitor: Failed to get response from the method.

dhruvibm, 2017-09-20:
# **STATUS**
9/20: Ready for merge review

closed 2017-09-26 by williamspatrick