openbmc/openbmc #2332: DHCP not correctly setup after update to new BMC image

geissonator, 2017-09-18:
```
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r5-0-g0b8d737"
```

After the update, all seemed to be ok except the mac address could not be found from the DHCP server.  On further investigation, it seems the network application is hitting errors and not properly setting up the default network config file for DHCP.

Because of this we end up using the /usr/lib/systemd/network/default.network  file which does not have the right DHCP defaults in it.

```
root@witherspoon:~# journalctl | cat | grep network
Sep 18 21:15:46 witherspoon kernel: Driver for 1-wire Dallas network protocol.
Sep 18 21:15:46 witherspoon kernel: netconsole: network logging started
Sep 18 21:15:54 witherspoon kernel[953]: [    2.130000] Driver for 1-wire Dallas network protocol.
Sep 18 21:15:54 witherspoon kernel[953]: [    7.720000] netconsole: network logging started
Sep 18 21:16:10 witherspoon systemd-networkd[1062]: Enumeration completed
Sep 18 21:16:10 witherspoon systemd-networkd[1062]: eth0: IPv6 enabled for interface: Success
Sep 18 21:16:10 witherspoon systemd-networkd[1062]: eth0: Gained carrier
Sep 18 21:16:11 witherspoon systemd-networkd[1062]: eth0: Gained IPv6LL
Sep 18 21:16:16 witherspoon systemd-networkd[1062]: eth0: DHCPv4 address 192.168.73.158/22 via 192.168.72.1
Sep 18 21:16:24 witherspoon systemd-networkd[1062]: eth0: Configured
Sep 18 21:16:43 witherspoon phosphor-network-manager[1133]: ConfigParser: Section not found
Sep 18 21:16:43 witherspoon phosphor-network-manager[1133]: The operation failed internally.
Sep 18 21:16:43 witherspoon phosphor-network-manager[1133]: Exception occured during getting of DHCP value
Sep 18 21:17:09 witherspoon systemd-networkd[1062]: eth0: Lost carrier
Sep 18 21:17:09 witherspoon systemd-networkd[1062]: eth0: DHCP lease lost
Sep 18 21:17:09 witherspoon systemd-networkd[1062]: eth0: Gained carrier
Sep 18 21:17:10 witherspoon systemd-networkd[1210]: Enumeration completed
Sep 18 21:17:10 witherspoon systemd-networkd[1210]: eth0: Gained IPv6LL
Sep 18 21:17:10 witherspoon phosphor-network-manager[1133]: ConfigParser: Section not found
Sep 18 21:17:10 witherspoon phosphor-network-manager[1133]: The operation failed internally.
Sep 18 21:17:10 witherspoon phosphor-network-manager[1133]: Exception occured during getting of DHCP value
Sep 18 21:17:11 witherspoon phosphor-network-manager[1133]: Timer expired
Sep 18 21:17:14 witherspoon systemd-networkd[1210]: eth0: DHCPv4 address 192.168.74.180/22 via 192.168.72.1
Sep 18 21:17:15 witherspoon phosphor-network-manager[1133]: ConfigParser: Section not found
Sep 18 21:17:15 witherspoon phosphor-network-manager[1133]: The operation failed internally.
Sep 18 21:17:15 witherspoon phosphor-network-manager[1133]: Exception occured during getting of DHCP value
Sep 18 21:17:15 witherspoon phosphor-network-manager[1133]: Timer expired
root@witherspoon:~# ls /etc/systemd/network/
root@witherspoon:~# 
root@witherspoon:~# cat /usr/lib/systemd/network/default.network 
[Match]
Name=eth*
[Network]
DHCP=ipv4
```

geissonator, 2017-09-18:
Kind of feels to me like in getDHCPValue() that the default should be true.  So if we get the exception trying to find the network config file, we default to DHCP?
```
345bool getDHCPValue(const std::string& confDir, const std::string& intf)
346{
347    bool dhcp = false;
348    // Get the interface mode value from systemd conf
349    //using namespace std::string_literals;
350    fs::path confPath = confDir;
351    std::string fileName = systemd::config::networkFilePrefix + intf +
352                           systemd::config::networkFileSuffix;
353    confPath /= fileName;
354
355    try
356    {
357        config::Parser parser(confPath.string());
358        auto values = parser.getValues("Network", "DHCP");
359        // There will be only single value for DHCP key.
360        if (values[0] == "true")
361        {
362            dhcp = true;
363        }
364    }
365    catch (InternalFailure& e)
366    {
367        log<level::INFO>("Exception occured during getting of DHCP value");
368    }
369    return dhcp;
370}
```

geissonator, 2017-09-18:
Seems related to discussion in https://github.com/openbmc/openbmc/issues/865#issuecomment-325756954

ratagupt, 2017-09-19:
@geissonator :- Systemd looks at first /etc/systemd/network/00-bmc-eth0.network...

ratagupt, 2017-09-19:
Seems new use case when we update from old code update(nonubi) to new code update(ubi based).
It wipes out all the data files.

ratagupt, 2017-09-19:
Since barellye this situation never occured that the network file wlll not be available after any code update hence this issue occured.

closed 2017-10-03 by geissonator