openbmc/openbmc #2333: If mac address is all 0's, code will still use it

geissonator, 2017-09-18:
Seen this a few times now.  The MAC address is all 0's in vpd and we somehow end up with that in our env variable.  Thought we had checking in place to avoid this.

```
00000100  00 42 31 08 00 00 00 00  00 00 00 00 50 46 03 00  |.B1.........PF..|
ethaddr=00:00:00:00:00:00
```

```
VERSION_ID="ibm-v1.99.10-0-r5-0-g0b8d737"
```

FYI @dkodihal @ratagupt

ngorugan, 2017-10-30:
I am not seeing the issue mentioned above.

I have done below given tests.

1. Updated MAC address in VPD(eeprom) with 000000000000(all zeros)
#Reading MAC address form VPD
root@witherspoon:~# /usr/bin/hexdump -C /sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a400.i2c-bus/i2c-11/11-0051/eeprom|grep ^00000100 |awk '{print $6$7$8$9$10$11}'
000000000000
2. Reading MAC address using ifconfig
root@witherspoon:~# ifconfig |grep HWaddr|awk '{print $NF}'|  sed "s/[:]//g"
70E284143364

3.Rebooted the system and read the MAC address using 'ifconfig' command. The value that  I got is always:70E284143364. Run this test couple of times.

@geissonator  Plz let me know If I am missing anything.

ratagupt, 2017-10-30:
@geissonator :- we can not set all zeros through set macaddress as mac address should be a local admin mac address.
@ngorugan :- you can verify it and paste the result.

ngorugan, 2017-10-30:
I have tried  couple of times.. and able to recreate the issue:

root@witherspoon:~#  /usr/bin/hexdump -C /sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a400.i2c-bus/i2c-11/11-0051/eeprom|grep ^00000100 |awk '{print $6$7$8$9$10$11}'
000000000000
root@witherspoon:~# ifconfig |grep HWaddr|awk '{print $NF}'| sed "s/[:]//g"
70E284143365
root@witherspoon:~#  fw_printenv ethaddr
ethaddr=00:00:00:00:00:00

ratagupt, 2017-10-30:
@ngorugan :- I figured it out that there is still a possibility where we see if the given mac is same as whatsoever is written in the vpd then we allow it. in that case networkd allows the all 0 to set it.
in that case our boot env variable output and the ip out put would not be in sync.
so you can write a condition in networkd where you should not allow the all zero mac address.
as underlying kernel doesn't allow all zero mac address.

amboar, 2017-10-30:
@ngorugan Please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

rfrandse, 2017-10-31:
<https://gerrit.openbmc-project.xyz/7690> Fix MAC Address validation
Resolves  #2333 If mac address is all 0's, code will still use it

closed 2017-11-06 by geissonator