openbmc/openbmc #2334: Move witherspoon power fault errors to org.open_power.witherspoon

spinler, 2017-09-19:
Currently they are xyz.openbmc_project, but since the repo is witherspoon specific we should make the errors witherspoon specific.

bjwyman, 2017-09-20:
Task List
- [ ] Should just be a simple `git mv` with a new witherspoon directory and re-run phosphor-logging tool.

bjwyman, 2017-09-20:
Do we want the namespace to change from:
`sdbusplus::xyz::openbmc_project::Power::Fault::Error`
to 
`sdbusplus::org::open_power::witherspoon::Power::Fault::Error`
etc.?

bjwyman, 2017-09-22:
# **STATUS**
**09/22**
- Today: No activity.
- Next: Wait for #1731 and #1732 and witherspoon-pseq-monitor changes to merge.
- Blockers: #1731, #1732, #1726.

rfrandse, 2017-10-19:
<https://gerrit.openbmc-project.xyz/7115> Update error YAML path
Resolves:  #2334 Move witherspoon power fault errors to org.open_power.witherspoon

closed 2017-10-26 by geissonator