openbmc/openbmc #2336: Fan Monitor: Handle MAX31785revA 0 RPM on speed change

msbarth, 2017-09-19:
A thermal requirement surfaced to handle the design issue within the MAX31785revA fan controller chip where the tachsel stops toggling to read both tach feedbacks for a counter-rotating rotor fan when the fan's target speed is changed. The tachsel begins to toggle again once the chip has determined the other rotor is within some range(determined by Maxim) of the requested target speed.

A configurable ability of the fan monitor application is needed where a given subset of fans are removed from being monitored when the entire subset is reading a provided value, in this case 0.
i.e.) Stop monitoring fan0_1, fan1_1, fan2_1, & fan3_1 when **all** of them report a 0 rpm

spinler, 2017-09-20:
[ ] - In a generic way, change fan monitor to not trust a 0 tach reading of certain tachs when other certain tachs are all 0.  Will involve yaml/parser/code updates.

rfrandse, 2017-10-02:
<https://gerrit.openbmc-project.xyz/6989> Add tach trust groups to witherspoon monitor yaml
Resolves  #2336 Fan Monitor: Handle MAX31785revA 0 RPM on speed change

closed 2017-10-11 by geissonator