openbmc/openbmc #2337: IPL: BMC Code update failure due to org.freedesktop.DBus.Error.FileNotFound prior to BMC reboot on w55 with 910.1738.20170919a

cdeadmin, 2017-09-20:


rfrandse, 2017-09-20:
<https://gerrit.openbmc-project.xyz/6775> phosphor-log-manager: Fix for deserialize cereal exception
Resolves  #2337 IPL: BMC Code update failure due to org.freedesktop.DBus.Error.FileNotFound prior to BMC reboot on w55 with 910.1738.20170919a

ojayanth, 2017-09-21:
Looks like known problem in un-clean reboot.

http://www.linux-mtd.infradead.org/faq/ubifs.html:
"Why is my file empty after an unclean reboot?
Zero-length files are a special case of corruption which happens when an application first truncates a file, then updates it. The truncation is synchronous in UBIFS, so it is written to the media straight away. But when the data are written, they go to the page cache, not to the flash media. So when an unclean reboot happens, the file becomes empty (truncated) because the data are lost."

ojayanth, 2017-09-21:
I was able to recreate this issue by continuously invoking the serialize() function ( Wrapper function for cereal::BinaryOutputArchive )  and killing  the application during this execution.

void serialize(const ElogList& list, const fs::path& dir)
{
  std::ofstream os(dir.c_str(), std::ios::binary);
  cereal::BinaryOutputArchive oarchive(os);
   oarchive(list);
}

When the application is running, output file size varies  but not necessarily increment as expected. 
Eg Data:
ls -l elogid
-rw-r--r--    1 root     root         28672 Sep 21 04:32 elogid
-rw-r--r--    1 root     root         16384 Sep 21 04:33 elogid
-rw-r--r--    1 root     root             0 Sep 21 04:33 elogid
-rw-r--r--    1 root     root         38852 Sep 21 04:33 elogid

This experiment proves that , an application crash in the middle of cereal archive operation can result in empty/truncated/corrupted files . 

De-serialization   of corrupted files leads to this failure.

ojayanth, 2017-09-21:
i partially fixed the issue using known solution  (http://www.linux-mtd.infradead.org/faq/ubifs.html#L_atomic_change) problem , and results looks good for the same test case.

ojayanth, 2017-09-22:
# **Status**
09/22: Workaround ode is Merged in GHE,  Waiting for same code merge in master.
             Discussed the proposed l fix in 09/21 design meeting.  Waiting for review comments from Brad. Also team decided to use new issue to track the actual fix.

closed 2017-09-22 by williamspatrick