openbmc/openbmc #2338: [xCAT] Is boot source interface changed?

xuweibj, 2017-09-20:
Version:
Host    Active(*)  IBM-witherspoon-sequoia-ibm-OP9_v1.19_1.30
BMC     Active     ibm-v1.99.10-0-r5-0-g0b8d737

I tried to use command ``curl -k -b cjar -X GET -H "Content-Type: application/json" https://x.x.x.x/xyz/openbmc_project/control/host0/boot_source`` as before to get boot source, got response ``path or object not found /xyz/openbmc_project/control/host0/boot_source``.
And I saw that the url ``xyz/openbmc_project/control/host0/boot`` has BootSource info.

Is the boot source interface changed? If changed, started from which version? Will be changed again in the future?

@whowutwut @zet809

geissonator, 2017-09-20:
@devenrao - can you answer this?  I know @dkodihal did some changes recently to support the "one time" concept but didn't think we changed the name of the variable?

vishwabmc, 2017-09-20:
It is now :

/xyz/openbmc_project/control/host0/boot <http://9.3.185.83:3000/#>

  * org.freedesktop.DBus.Peer <http://9.3.185.83:3000/#>
  * org.freedesktop.DBus.Introspectable <http://9.3.185.83:3000/#>
  * org.freedesktop.DBus.Properties <http://9.3.185.83:3000/#>
  * xyz.openbmc_project.Control.Boot.Mode <http://9.3.185.83:3000/#>
      o BootMode = xyz.openbmc_project.Control.Boot.Mode.Modes.Regular
        <http://9.3.185.83:3000/#>
  * xyz.openbmc_project.Control.Boot.Source <http://9.3.185.83:3000/#>
      o BootSource =
        xyz.openbmc_project.Control.Boot.Source.Sources.Default
        <http://9.3.185.83:3000/#>

tomjoseph83, 2017-09-20:
@xuweibj https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Boot/Source.interface.yaml

zet809, 2017-09-21:
hi @vishwabmc , so the interface /xyz/openbmc_project/control/host0/boot will be used from now on from Firmware level OP910_1738A? Why it is changed? 
Is it possible to change it in the future? How do we know if it is changed in the future? Thx!

tomjoseph83, 2017-09-21:
@zet809 The object paths and interface are well defined, we do not foresee any change.This is probably an exception. The users/stakeholders would be informed next time( eg: mailing list) and the documentation would be elaborated the docs section.

whowutwut, 2017-09-25:
@tomjoseph83 which mailing list should we be on to know about these kind of changes and how far in advance would a notification happen?  

So after that PR was merged, we were now broken on all our machines with FW code < 1738...  which lots of our internal machines are at.... and even machines coming off mfg...    so had to provide yet another workaround patch for supporting both endpoints..

dkodihal, 2017-09-25:
> hi @vishwabmc , so the interface /xyz/openbmc_project/control/host0/boot will be used from now on from Firmware level OP910_1738A

This is not accurate. While the above object path exists, it's not enabled by default. See my comment above @zet809.

vishwabmc, 2017-09-26:
@dkodihal not sure I understand. I just mentioned the object path that implemented the interfaces shown above. Missing thing was mentioning permanent. yea !

Probably, this discussion can get into the setting repo docs ?

tomjoseph83, 2017-10-02:
@xuweibj Please close this issue, if the question has  been answered sufficiently.

xuweibj, 2017-10-09:
Have got correct information, close this issue. Thanks @tomjoseph83 .

closed 2017-10-09 by xuweibj