openbmc/openbmc #2341: Sprint 17.10.04 - Committed

zahrens, 2017-09-20:


dkodihal, 2017-10-03:
***Squad 1 sprint plan***

**Deepak**
#2317 External Event Notification : define API
#2318 External Event Notification : Implement API

**Dhruv**
#2319 Events : persist using Cereal
#2321 Install gevent-websocket to the image

**Jayanth and Devender**
#2037 Enable support for usecase specfic bash script invocation
#2351 dreport plugin support  

**Tom**
#2380 Handle new eSEL from host which will contain OCC telemetry data

**Nagaraju**
#2378 Remove org.openbmc.* from REST server
#2381 Handle more then 100 error logs

**Ratan**
- Wrap up #2254 Implement event logging mechanism
#2399 Yaml changes to create events

Need help with
#1725 add GPUs to inventory
#2398 Allow in band IPMI to enable/disable automatic reboots

spinler, 2017-10-04:
***Squad 2 Sprint Plan***

**Matt S**

**Matt B**

**Gunnar**

**Brandon**

**Eddie**

anoo1, 2017-10-04:
***Squad 3 Sprint Plan***

**Adriana**
#1627 Save the systemd journal to a persistent storage
#2283 Deleting the current BMC version results in boot failure

**Saqib**
#1944 BMC: Block BMC reboot during u-boot update
#2323 At build time, calculate the version id of the BMC ubi volumes

**Michael**
#2389 No API to clear GUARD, leading to bricked systems
#2416 Need a method to create an internal error in error logs

**Chris**

**Charlie**

closed 2017-10-26 by rfrandse