openbmc/openbmc #2343: FSI bus errors encountered during FIFO driver operations

cbostic, 2017-09-20:
Murulidhar Nataraju provided the following error data when he hits this FIFO driver related issue:

```
I have copied the required data below... can you help me debug why read is not getting completed...
I am hitting POLL ERR when I am pooling for he read event.

dmesg: (I see bus access failure)

[56442.640000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008
[56442.660000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008
[56935.250000] sbefifo 00:00:00:06: Fatal bus access failure: -5
[56935.250000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
"dmsg_2" 187L, 13473C                                                                                                                                                       

Full Dmsg file is at : /afs/rch/usr5/murunata/public/dmsgFile

STRACE Output:

root@witherspoon:/tmp# bin/strace -e trace=open,write,read,poll ./sbei_inter
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0,\333\6\0004\0\0\0"..., 512) = 512
open("/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240\322\0\0004\0\0\0"..., 512) = 512
open("/lib/tls/v6l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/v6l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 h\1\0004\0\0\0"..., 512) = 512
open("/lib/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\220=\0\0004\0\0\0"..., 512) = 512
open("/dev/sbefifo1", O_RDWR|O_NONBLOCK) = 3
poll([{fd=3, events=POLLOUT|POLLERR}], 1, -1) = 1 ([{fd=3, revents=POLLOUT}])
write(3, "\4\0\0\0\0\0\242\1\0\0\0\0\0 \20\7", 16) = 16                                                      ==================> I see the write passed 
poll([{fd=3, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=3, revents=POLLERR}])=================> This is line 76 in the image below
write(1, "Hit a POLL Error errno:0x0\n", 27Hit a POLL Error errno:0x0                       =================>  This is line 87 in the image below
) = 27```

cbostic, 2017-09-20:
These particular FSI bus errors are not in the same time vicinity of the sbefifo driver error message:

[56442.640000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008
[56442.660000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008

Probably not related.  Will need to gather a full bus trace to identify reasons for the sbefifo -5 error.

cbostic, 2017-10-12:
Interesting findings today:
1.  Rate of CRC errors leading to poll fail appears to depend on what system is being used.
    w5: never,  zaius2: often.
2.  Once CRC error occurs the CFAM in question can be busy for a while before further FSI transactions process successfully.

cbostic, 2017-10-16:
Marking as 'Phase 7'

cbostic, 2017-10-20:
Have a fix ready for review for this issue.

The issue is due to a time window where the P9 FSI slave is not always responding to BMC FSI accesses done via the sbe_fifo device driver.  This window is approximately 80 seconds after FSI scan finishes during a power up and lasts for about 9 seconds.  After this all sbe_fifo device driver polling works without any issues.   

A BREAK is issued by the FSI device driver to recover links that get into a bad state, however, during this fail window a single one may not always be enough.  I've found via several test runs on a few different systems that typically 2 or 3 BREAKs will take care of the matter.  Occasionally 4 are needed.

Adding a change to FSI driver to retry recovery BREAKs up to 5 times before giving up and returning an error to the sbe_fifo client driver (other clients will see same behavior).

amboar, 2017-10-22:
Do we actually know what the cause is? Why is retrying some number of times the right fix? Are we sure that by the time we're done executing some number of breaks we haven't passed the "fail window"? How was the temporal location and duration of the fail-window measured? What is going on in the host when this occurs? Are there any obvious causal relationships between host initialisation and this fail-window?

cbostic, 2017-10-23:
>>  Do we actually know what the cause is?
  No, at this point all I can tell is that the CFAM FSI slave flags errors during normal transactions and that attempting to pull data out of slave to narrow down fail cause itself causes further bus access failures.

>> Why is retrying some number of times the right fix?
Typically a BREAK resets the FSI slave.  But, I've seen cases in test runs where one BREAK is not enough.  You should be able to access base slave registers after a BREAK and in the fail mode mentioned that isn't always happening.  A repeated break gets out of the fail mode.

>> Are we sure that by the time we're done executing some number of breaks we haven't passed the "fail window"?
  After recovering link this way I've seen cases where we again hit this fail mode requiring multiple breaks.

>> How was the temporal location and duration of the fail-window measured?
Simply measured roughly via debug time stamps when FSI scan begins signalling approximate beginning of power on to when the SBE FIFO accesses via FSI bus start randomly failing.   End of fail window was measured by noting last fail time and comparing to several test runs...  These are not precise time values just a general time line.

rfrandse, 2017-10-27:
defect transferred to hostboot

closed 2017-10-27 by rfrandse