openbmc/openbmc #2344: Patching host firmware doesn't work

cdeadmin, 2017-09-20:


sjitindarsingh, 2017-10-10:
Looks like it's impossible with vpnor to ever read the last non-block size aligned bytes of a file.

Say a file size mod block size is 100 bytes.

Then in mboxd_flash_virtual.cpp:181:
rc = std::min(partitionInfo->data.actual - diffOffset, size);

And in mboxd_windows.c:638:
cur->size = rc;

And finally in mboxd_msg.c:379:
put_u16(&resp->args[2], context->current->size >>
context->block_size_shift);

We lose that last 100 bytes when we perform the shift (essentially aligning down the size).

What we end up with is the file size being truncated to the block size when it is reported back to the host.
Really we want to be rounding the window size up to the nearest block size.

sjitindarsingh, 2017-10-10:
Patch posted here:

https://gerrit.openbmc-project.xyz/#/c/7096/

amboar, 2017-10-11:
@sjitindarsingh Given the nature of the bug I'd like some tests to make sure we don't regress going forward. What's our best route for getting them written?

bradbishop, 2017-10-11:
https://github.com/openbmc/openbmc/issues/2452 opened for test case.

closed 2017-10-11 by geissonator