openbmc/openbmc #2346: Fan Control: Handle configurable actions on dependent application failures

msbarth, 2017-09-20:
Fan control should react to dependent application failures/crashes by a user configured action. Subscribing to the `NameOwnerChanged` signal should allow the application to know when an application its dependent on for input to fan control fails/crashes/exits. Upon the application exit, a configurable action should be performed based on specific application that no longer exists since the properties it was updating on the bus will still exist and remain unchanged.

msbarth, 2017-09-21:
- [ ] Create signal functor for `NameOwnerChanged` signals
- [ ] Add application service name cache for subscribed `NameOwnerChanged` signals
- [ ] Create handler to update application service name cache
- [ ] Store application service name on init in cache, calling action when defined service name is not found
- [ ] Add action to set fans to defined speed upon application failure
- [ ] Add action to terminate fan control application upon application failure (tripping watchdog)
- [ ] Modify parser to support `signals` section listing available signals for events to subscribe to
- [ ] Subscribe to `NameOwnerChanged` signals for hwmon sensor application instances
- [ ] Subscribe to `NameOwnerChanged` signals for OCC application instance

msbarth, 2017-10-11:
Starting end-to-end testing of changes with the terminate fan control action upon service fail of the max31785.

msbarth, 2017-10-24:
All service fail actions added and verified. Fixing bug in object mapper to provide the well-known bus name for `NameOwnerChanged` to be handled as intended. Also working on a necessary update to init the target speed on application start from dbus.

rfrandse, 2017-10-31:
<https://gerrit.openbmc-project.xyz/7678> wspoon: Handle dependent application fails
Resolves  #2346 Fan Control: Handle configurable actions on dependent application failures