openbmc/openbmc #2347: Failed to start SYNC system/chassis/motherboard/bmc/ethernet.

gkeishin, 2017-09-21:
Seen on multiple systems on AES, where the system reported failure. On closer look

On reboot

```
Sep 21 17:40:29 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Main process exited, code=exited, status=1/FAILURE
Sep 21 17:40:29 witherspoon systemd[1]: Failed to start Sync system/chassis/motherboard/bmc/ethernet.
Sep 21 17:40:29 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Unit entered failed state.
Sep 21 17:40:29 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Failed with result 'exit-code'.
```
`FW info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"
root@witherspoon:~#
```

- [x] Journald 
[MAC_sync_failure.txt](https://github.com/openbmc/openbmc/files/1322336/MAC_sync_failure.txt)

ratagupt, 2017-09-21:
If there is change in mismatch in MAC between system mac and inventory it always work..In first go it fails and printed the above failures and in second go it synchs.
```Sep 18 14:09:26 witherspoon sync_inventory_items.py[1412]: Unable to get the network object
Sep 18 14:09:26 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 14:09:26 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Unit entered failed state.
Sep 18 14:09:26 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Failed with result 'exit-code'.
Sep 18 14:10:11 witherspoon sync_inventory_items.py[1622]: Inventory MAC=70:e2:84:14:23:9f,System MAC=00:00:00```

geissonator, 2017-09-30:
I did a test where I changed my ethaddr boot env variable and did a reboot on a static IP configured system.  I recreated this fail.  I think we get lucky on most systems and restart the synch service because a power_policy of RESTORE causes an additional power off once we reach Ready state.  If you're system doesn't have RESTORE then we wont restart the synch service and you'll end up with a mismatch in mac.  There's a huge time difference here between when the phosphor network manager starts and the inventory synch starts, so a race condition there doesn't seem obvious unless the network manager has some large delay in creating it's objects after starting.
```
[witherspoon]# tracf | grep -e "Sync system" -e "Phosphor Network Manager" -e "Unit entered failed" -e "sync_inventory_items.py" -e mapper
Sep 30 14:23:04 witherspoon systemd[1]: Starting Phosphor Network Manager...
Sep 30 14:23:39 witherspoon phosphor-mapper[965]: ObjectMapper startup complete.  Discovery in progress...
Sep 30 14:23:41 witherspoon systemd[1]: Started Phosphor Network Manager.
Sep 30 14:23:59 witherspoon phosphor-mapper[965]: ObjectMapper discovery complete
Sep 30 14:23:59 witherspoon systemd[1]: Starting Phosphor Network Manager...
Sep 30 14:24:06 witherspoon systemd[1]: Started Phosphor Network Manager.
Sep 30 14:24:22 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/uuid...
Sep 30 14:24:22 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/bmc/ethernet...
Sep 30 14:24:29 witherspoon systemd[1]: Started Sync system/chassis/motherboard/uuid.
Sep 30 14:24:31 witherspoon sync_inventory_items.py[1241]: Unable to get the network object
Sep 30 14:24:31 witherspoon systemd[1]: Failed to start Sync system/chassis/motherboard/bmc/ethernet.
Sep 30 14:24:31 witherspoon systemd[1]: obmc-sync-inventory-item@system-chassis-motherboard-bmc-ethernet.service: Unit entered failed state.
Sep 30 14:24:52 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/uuid...
Sep 30 14:24:52 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/bmc/ethernet...
Sep 30 14:24:59 witherspoon sync_inventory_items.py[1325]: Inventory MAC=70:e2:84:14:16:6c,System MAC=70:e2:84:14:16:99
Sep 30 14:24:59 witherspoon systemd[1]: Started Sync system/chassis/motherboard/uuid.
Sep 30 14:25:01 witherspoon systemd[1]: Started Sync system/chassis/motherboard/bmc/ethernet.
```

geissonator, 2017-09-30:
I changed the mac, and did a reboot and just kept running 
```
date && busctl introspect xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 | cat && busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObject sas "/xyz/openbmc_project/network/eth0" 1 "xyz.openbmc_project.Network.EthernetInterface" 
```

Def see a big delay between when the object is directly available and when mapper has it loaded so it does seem like the right solution is what @ratagupt was going for of putting a mapper dependency in the inventory service file.

gkeishin, 2017-10-16:
@ratagupt anything new on this one. I keep seeing this on today Master, on boot to standby.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-154"
VERSION_ID="v1.99.10-154-g63d4f19"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-154"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-97988a39b46d73304a5697b8d3a5cf08da287350 #1 Tue Oct 10 05:30:30 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald log 
[OBMC_SYNC_error.txt](https://github.com/openbmc/openbmc/files/1386537/OBMC_SYNC_error.txt)

dhruvibm, 2017-12-14:
# **Status**
Not recreating need to close

gkeishin, 2017-12-15:
I ll check on it today and close it up it doesnt show up. Thanks

gkeishin, 2018-01-16:
Not seen in MAC configured system anymore.

closed 2018-01-16 by gkeishin