openbmc/openbmc #2349: phosphor-gpio-monitor ABRT

shenki, 2017-09-22:
doing `obmcutil poweron` running openbmc master as of v1.99.10-19-g319ec41 on Plametto:

```
Sep 22 04:42:35 palmetto phosphor-gpio-monitor[1236]: Failed to open device path
Sep 22 04:42:35 palmetto phosphor-gpio-monitor[1236]: The operation failed internally.
Sep 22 04:42:35 palmetto phosphor-gpio-monitor[1236]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Sep 22 04:42:36 palmetto phosphor-gpio-monitor[1236]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Sep 22 04:42:35 palmetto systemd[1]: phosphor-gpio-monitor@checkstop.service: Main process exited, code=dumped, status=6/ABRT
Sep 22 04:42:35 palmetto systemd[1]: phosphor-gpio-monitor@checkstop.service: Unit entered failed state.
Sep 22 04:42:35 palmetto systemd[1]: phosphor-gpio-monitor@checkstop.service: Failed with result 'core-dump'.
```

vishwabmc, 2017-09-22:
@shenki GPIO monitor uses `/dev/input/by-path/platform-gpio-keys-event` and afaik, this was witherspoon only path.

@eddiejames know anything about GPIO input key for Palmetto and Barreleye ?

kachini, 2017-11-24:
any news on this issue? just checked the latest code and checkstop service still fails

mine260309, 2017-11-27:
This issue impact the RAS test on P8 systems.

The root cause is probably due to missing `checkstop` gpio-key, and thus it fails to open the device.

For P8 systems:
1. `checkstop` needs to be defined in dts
2. Iif the key is different from P9 (GPIOJ2), `checkstop-monitor.bb` shall be appended to override the default `KEY` value.

kachini, 2017-11-27:
AFAIK this was fixed on other P8 platforms, wondering if someone pushed the same for barreleye

mine260309, 2017-11-28:
I will push the changes for P8 systems.

rfrandse, 2017-11-28:
<https://gerrit.openbmc-project.xyz/8077> meta-p8 systems: Fix phosphor-gpio-monitor crash
Fixes  #2349  phosphor-gpio-monitor ABRT

mine260309, 2017-11-28:
* dts changes: http://patchwork.ozlabs.org/patch/842049/
* openbmc change: https://gerrit.openbmc-project.xyz/#/c/8077/

closed 2017-12-12 by geissonator