openbmc/openbmc #2351: dreport plugin support

ojayanth, 2017-09-22:
Non-recipe related changes tracked here.
 in https://github.com/openbmc/openbmc/issues/2037

ojayanth, 2017-10-05:
- [ ] dreport utility function support
logging ( info/important/summary/warning).
Error handling
Saving contents
Getting dump context Configuration parameters

- [ ] restructure the existing  dreport  code to support
CLI infrastructure
Initializing the  dump type config parameters
Invoking data collection based on the dump type
compression and copy  into permanent location
Error management

- [ ] Create individual bash scripts for the existing data collection functions.

rfrandse, 2017-10-16:
<https://gerrit.openbmc-project.xyz/7358> Change dreport source location to tools/dreport.d/dreport
Resolves  #2351 dreport plugin support

closed 2017-10-24 by geissonator