openbmc/openbmc #2352: Tooling to cherry pick specific changes into releases and branches

geissonator, 2017-09-22:
Once we get close to shipping systems, we'll need to start locking down internal branches and tags.  Would like some tooling (via this story) where a user could point to specific patches (either merged in github, or un-merged changes in gerrit) and a target branch which would:
- Cherry pick the change(s)
- Generate patch(es) based on input tag
- Copy patch to appropriate dir, update corresponding .bb file to use patch

(this is all via the openbmc/openbmc tree)

Also do the following:
- Take the input tag, take the new tag, checkout the tag, apply the fixes, create the new tag

amboar, 2017-10-03:
Does yocto already provide something in this regard? This process is applicable to all the tools and applications it provides recipes for.

geissonator, 2018-01-17:
Could also use this to verify changes in sub-repo's in CI (i.e. do full openbmc build and validate)

rfrandse, 2018-08-16:
moved to https://github.com/ibm-openbmc/dev/issues/68

closed 2018-08-16 by rfrandse