openbmc/openbmc #2358: witherspoon openpower-occ-control: Opening OCC device failed

gkeishin, 2017-09-25:
```
Sep 15 16:26:12 witherspoon openpower-occ-control[1286]: Opening OCC device failed.
Sep 15 16:26:12 witherspoon openpower-occ-control[1286]: terminate called after throwing an instance of 'sdbusplus::org::open_power::OCC::Device::Error::OpenFailure'
Sep 15 16:26:12 witherspoon openpower-occ-control[1286]:   what():  org.open_power.OCC.Device.Error.OpenFailure: Opening OCC device failed.
Sep 15 16:26:12 witherspoon systemd[1]: org.open_power.OCC.Control.service: Main process exited, code=dumped, status=6/ABRT
Sep 15 16:26:12 witherspoon systemd[1]: org.open_power.OCC.Control.service: Unit entered failed state.
Sep 15 16:26:12 witherspoon systemd[1]: org.open_power.OCC.Control.service: Failed with result 'core-dump'.
Sep 15 16:26:12 witherspoon systemd[1]: Started Phosphor Time Manager daemon.
Sep 15 16:26:13 witherspoon systemd[1]: org.open_power.OCC.Control.service: Service hold-off time over, scheduling restart
```

Seen during OCC testing

geissonator, 2017-09-25:
I was able to recreate this with master as well.  The issue is that we shouldn't call op-occ-enable@.service until after the fsi scan services have run

gkeishin, 2017-09-25:
Closing this one as we are bridging with the CQ. Thanks

closed 2017-09-25 by gkeishin