openbmc/openbmc #2359: HOST.Updater core dump, PNOR images in software/enumerate disappear

charleshofer, 2017-09-25:
Steps to reproduce:

1. Update to -r7
2. Do a PNOR code update

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"
```

```Sep 25 20:03:12 witherspoon kern.err openpower-update-manager[2937]: Failed to read pnorTOC
Sep 25 20:03:12 witherspoon kern.err openpower-update-manager[2937]: Error in reading file
Sep 25 20:03:12 witherspoon daemon.info openpower-update-manager[2937]: terminate called after throwing an instance of 'std::ios_base::failure'
Sep 25 20:03:12 witherspoon daemon.info openpower-update-manager[2937]:   what():  basic_ios::clear
Sep 25 20:03:12 witherspoon daemon.warn systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=dumped, status=6/ABRT```

[journal.txt](https://github.com/openbmc/openbmc/files/1330602/journal.txt)

saqibkh, 2017-09-26:
The issue originated with a corrupted pnor volume as pnor-ro-<versionID> was empty (no pnor.toc).
We need to handle this issue more gracefully then causing a core-dump.

rfrandse, 2017-09-26:
<https://gerrit.openbmc-project.xyz/6881> Delete the pnor version that doesn't have a valid pnor.toc
Resolves  #2359 HOST.Updater core dump, PNOR images in software/enumerate disappear

closed 2017-10-02 by geissonator