openbmc/openbmc #2360: witherspoon openpower-occ-control: Opening OCC device failed

cdeadmin, 2017-09-25:


geissonator, 2017-09-25:
On a BMC reset, with the host up, the openpower-occ-control application will fail to start because it tries to do the occ bind before the fsi scan's have completed.  Need to add a dependency to the op-occ-enable.service to not run until the scans are done.  @spinler has some new code coming in https://gerrit.openbmc-project.xyz/#/c/6748/ that will be removing the hub-scan service so I'll only set the main fsi-scan service as a dependency.

rfrandse, 2017-09-25:
<https://gerrit.openbmc-project.xyz/6861> Ensure occ active is sent after fsi scan
Resolves  #2360 witherspoon openpower-occ-control: Opening OCC device failed

geissonator, 2017-09-29:
# **STATUS**
9/29 - Fix up and tested, waiting on some other changes to get in ahead of it.

closed 2017-10-04 by geissonator