openbmc/openbmc #2361: 1738 FW code on Witherspoon isn’t setting up the MAC correctly on the BMC if it’s DHCP

moralesj, 2017-09-25:
BMC is DHCP on a Redbud planar.
After successfully running the update_mac.py script, the MAC address was written to the VPD on the 1736 code.  Once I ran the old-new-code-update.sh script to update to 1738 the MAC's on eth0 of the BMC kept changing after a BMC reboot.  

The B1 keyword is still written with the correct MAC

00000100  00 42 31 06 70 e2 84 14  29 0c 00 00 50 46 03 00  \|.B1.p...)...PF..\|
--

The machine's BMC console can be accessed using the following adress
telnet 9.41.164.39 2006

geissonator, 2017-09-25:
MAC address looks ok, but will need @ratagupt to comment on why it's not getting a DHCP address.  I assume you're sure there's a DHCP server that will provide an IP on this network?

```
Sep 25 19:35:25 witherspoon systemd[1]: Started Read OpenPOWER-format VPD from EEPROM.
Sep 25 19:35:25 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/uuid...
Sep 25 19:35:25 witherspoon systemd[1]: Starting Sync system/chassis/motherboard/bmc/ethernet...
Sep 25 19:35:32 witherspoon systemd[1]: Started Sync system/chassis/motherboard/uuid.
Sep 25 19:35:34 witherspoon sync_inventory_items.py[1267]: ## Error: "ethaddr" not defined
Sep 25 19:35:34 witherspoon sync_inventory_items.py[1267]: Inventory MAC=70:e2:84:14:29:0c,System MAC=
Sep 25 19:35:36 witherspoon avahi-daemon[1005]: Interface eth0.IPv6 no longer relevant for mDNS.
Sep 25 19:35:36 witherspoon systemd-networkd[992]: eth0: Lost carrier
Sep 25 19:35:36 witherspoon avahi-daemon[1005]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::4892:49ff:fef9:d07b.
Sep 25 19:35:36 witherspoon avahi-daemon[1005]: Withdrawing address record for fe80::4892:49ff:fef9:d07b on eth0.
Sep 25 19:35:36 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Sep 25 19:35:36 witherspoon kernel[1006]: [  102.820000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
Sep 25 19:35:36 witherspoon kernel: 8021q: adding VLAN 0 to HW filter on device eth0
Sep 25 19:35:36 witherspoon kernel[1006]: [  103.020000] 8021q: adding VLAN 0 to HW filter on device eth0
Sep 25 19:35:36 witherspoon systemd-networkd[992]: eth0: Gained carrier
Sep 25 19:35:36 witherspoon systemd[1]: Stopping Network Service...
Sep 25 19:35:36 witherspoon systemd[1]: Stopped Network Service.
Sep 25 19:35:36 witherspoon systemd[1]: Starting Network Service...
Sep 25 19:35:37 witherspoon systemd[1]: Started Sync system/chassis/motherboard/bmc/ethernet.
Sep 25 19:35:37 witherspoon systemd[1]: Started Wait for /org/openbmc/settings/host0.
Sep 25 19:35:37 witherspoon systemd[1]: Started Phosphor Inband IPMI.
Sep 25 19:35:37 witherspoon systemd[1]: Started Wait for /org/openbmc/managers/System.
Sep 25 19:35:37 witherspoon systemd[1]: Starting Phosphor Reset Button0...
Sep 25 19:35:37 witherspoon systemd[1]: Starting Phosphor BMC State Manager...
Sep 25 19:35:38 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
Sep 25 19:35:38 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Sep 25 19:35:37 witherspoon kernel[1006]: [  104.110000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
Sep 25 19:35:37 witherspoon systemd[1]: Starting Phosphor Power0 Control...
Sep 25 19:35:37 witherspoon kernel[1006]: [  104.110000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
Sep 25 19:35:37 witherspoon systemd[1]: Starting Phosphor Chassis State Manager...
Sep 25 19:35:37 witherspoon avahi-daemon[1005]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::72e2:84ff:fe14:290c.
Sep 25 19:35:37 witherspoon systemd[1]: Starting Phosphor Power Button0...
Sep 25 19:35:37 witherspoon avahi-daemon[1005]: New relevant interface eth0.IPv6 for mDNS.
Sep 25 19:35:38 witherspoon systemd-networkd[1278]: eth0: Gained IPv6LL
Sep 25 19:35:37 witherspoon avahi-daemon[1005]: Registering new address record for fe80::72e2:84ff:fe14:290c on eth0.*.
Sep 25 19:35:38 witherspoon systemd-networkd[1278]: Enumeration completed
Sep 25 19:35:38 witherspoon systemd-timesyncd[946]: Network configuration changed, trying to establish connection.
Sep 25 19:35:38 witherspoon systemd[1]: Started Network Service.
```

```
root@witherspoon:~# cat /etc/systemd/network/00-bmc-eth0.network 
root@witherspoon:~# 
```

moralesj, 2017-09-25:
I disconnected the BMC from the DHCP server for the mean time 
 
MAC addresses were changing every time the BMC rebooted...

root@witherspoon:~# ifconfigeth0      Link encap:Ethernet  HWaddr E6:43:D6:4F:B7:E2             inet addr:9.41.168.136  Bcast:9.41.171.255  Mask:255.255.252.0          inet6 addr: fe80::e443:d6ff:fe4f:b7e2%1204106128/64 Scope:Link          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1          RX packets:2651 errors:0 dropped:0 overruns:0 frame:0          TX packets:134 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1000           RX bytes:189022 (184.5 KiB)  TX bytes:16687 (16.2 KiB)          Interrupt:20
--

POWER CYCLE BMC 

root@witherspoon:~# ifconfigeth0      Link encap:Ethernet  HWaddr 4A:92:49:F9:D0:7B            inet6 addr: fe80::4892:49ff:fef9:d07b%1204106128/64 Scope:Link          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1          RX packets:5829 errors:0 dropped:0 overruns:0 frame:0          TX packets:173 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1000           RX bytes:429548 (419.4 KiB)  TX bytes:16591 (16.2 KiB)          Interrupt:20

Uma was on the system and stated

clearing ethaddr from fw_setenv fixed the issue
so probably that's the reason MAC's are correct at the moment

prkatti1, 2017-09-26:
I tried this on r7 release, I did not get this issue.
I tried 5 times reboot, MAC address is persistent.

mdmillerii, 2017-09-26:
> Uma was on the system and stated
>> clearing ethaddr from fw_setenv fixed the issue

We need to find out what the old value of `ethaddr` was.  Perhaps it was set but not to a valid address, which would cause the stated behavior.

moralesj, 2017-10-03:
SO....Andrew and Adriana were working on this issue, I heard some chatter, but just wanted to have some confirmation on here.

moralesj, 2017-10-03:
I'll just put a chat I had with Andrew G about the defect 

> JOSE E. M...
9:56:54 AM
HI andrews...yesterday you and Adriana were looking at my system who's DHCP kept on changing  
9:56:56 AM
1738 FW code on Witherspoon isn’t setting up the MAC correctly on the BMC if it’s DHCP #2361  
9:57:26 AM
do you guys need to hold the system in debug, or can i take it back ..if you need it, that's cool just want to know  

Andrew J....
9:58:14 AM
no, we don't need anymore, we figured out the issue with your system, we've got a fix coming  

JOSE E. M...
9:58:36 AM
ah ok, on what level of driver will it be coming out  

Andrew J....
9:59:31 AM
probably 1740, it's not in yet though, working the weekend on it  

JOSE E. M...
9:59:40 AM
k thank you!  

Andrew J....
9:59:46 AM
I think we probably fixed that system though, so you shouldn't see it  

JOSE E. M...
9:59:58 AM
ok cool  

Andrew J....
10:00:29 AM
you can basically run that command @anoo and I ran from the slack channel if you hit on other systems to force a read from vpd of the mac  

> 

>

geissonator, 2017-10-03:
Fix coming in https://gerrit.openbmc-project.xyz/#/c/6645/

rfrandse, 2017-10-04:
<https://gerrit.openbmc-project.xyz/6645> Inventory: Make changes for sync inventory for witherspoon
Resolves  #2361 1738 FW code on Witherspoon isn’t setting up the MAC correctly on the BMC if it’s DHCP

closed 2017-10-04 by geissonator

marksmit-ibm, 2017-10-06:
I verified w/ Andrew this is in R11 (aka Witherspoon 1740A firmware)