openbmc/openbmc #2362: witherspoon-pfault-analysis: Make use of the State.Shutdown.ThermalEvent.GPU and State.Shutdown.Power.Fault errors

spinler, 2017-09-25:
Need to start using these 2 new errors in phosphor-dbus-interfaces instead of the current ones in the witherspoon-pfault-analysis repository

rfrandse, 2017-10-10:
<https://gerrit.openbmc-project.xyz/7200> Use the phosphor-dbus-interfaces shutdown error
Resolves  #2362 witherspoon-pfault-analysis: Make use of the State.Shutdown.ThermalEvent.GPU and State.Shutdown.Power.Fault errors

rfrandse, 2017-11-28:
Nice to have. Moving to Phase 8 to revisit at another time.

geissonator, 2017-12-13:
#2705 should help here

rfrandse, 2017-12-13:
<https://gerrit.openbmc-project.xyz/7200> Use the phosphor-dbus-interfaces shutdown error
Resolves  #2362 System-pfault-analysis: Make use of the State.Shutdown.ThermalEvent.GPU and State.Shutdown.Power.Fault errors

spinler, 2018-02-20:
This still can't work due to how this would require a local elog-errors.hpp and the one from the include path being included in the same file, and then there are symbol conflicts.  As it's not like I would be completely removing repository local errors anyway, I'm just going to give up on this.

closed 2018-02-20 by spinler