openbmc/openbmc #2363: BMC crash with kernel trace after soft power off

bjwyman, 2017-09-25:
I am not quite certain what lead up to this, but it might be related to attempts to test power supply over temperature faults.

The host was in Quiesce state for quite some time. After final success with testing the over temperature power supply fault, some kind of request for a power off came in, followed by some kernel stack trace looking information, and then the connection to the BMC was lost.

I had `journalctl -f&` running during my testing, so I do have that information captured.
[soft_power_off_crash.txt](https://github.com/openbmc/openbmc/files/1330983/soft_power_off_crash.txt)

bjwyman, 2017-09-25:
```root@yl13ufbmc:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.9"
VERSION_ID="ibm-v1.99.9-0-r4-0-gf0746a0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.9"
BUILD_ID="ibm-v1.99.9-0-r4"
root@yl13ufbmc:~# ```

amboar, 2017-09-25:
@eddiejames is working on resolving this, need to find all the duplicates of this issue

bjwyman, 2017-09-25:
@eddiejames indicated this is a duplicate of #2345.

geissonator, 2017-09-25:
Closing as dup

closed 2017-09-25 by geissonator