openbmc/openbmc #2364: Power supply PMBus files missing on some systems

bjwyman, 2017-09-25:
I have seen this more than once, most recently seen on a system with STC level power supplies (the latest).
The status0_mfr and status0_cml files in /sys/kernel/debug/pmbus/hwmonX are missing.
Sending i2cget, one can get those values from the power supply, and also see that the STATUS_WORD response is 0x0002. Reading the STATUS_CML shows 0x80 (invalid/unsupported command). It is not clear what leads to that condition, or why that causes the ibm-cffps device driver to cause the two files mentioned to be missing.

bjwyman, 2017-09-25:
```
# #Get ON_OFF_CONFIG mode:                                       
# i2cget -f -y 3 0x69 0x02 b                                     
0x15                                                                             
# i2cget -f -y 3 0x68 0x02 b                                     
0x15                                                                             
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get SYS_CONFIG (Fault Indicator State of Amber LED)          
# # bxxxxx000 - PSU not faulted. LED off.                        
# # bxxxxx000 - PSU faulted. LED on solid.                       
# # bxxxxxxx1 - LED blink.                                       
# i2cget -f -y 3 0x69 0xda b                                     
0x00                                                                             
# i2cget -f -y 3 0x68 0xda b                                     
0x00                                                                             
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get power supply FRU numbers (ASCII)                         
# i2cget -f -y 3 0x69 0x9a i 13; i2cget -f -y 3 0x68 0x9a i 13   
13: 0x0c 0x30 0x31 0x4b 0x4c 0x38 0x37 0x30 0x20 0x20 0x20 0x20 0x20             
13: 0x0c 0x30 0x31 0x4b 0x4c 0x38 0x37 0x30 0x20 0x20 0x20 0x20 0x20             
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get power supply part numbers (ASCII)                        
# i2cget -f -y 3 0x69 0x9b i 13; i2cget -f -y 3 0x68 0x9b i 13   
13: 0x0c 0x30 0x31 0x4b 0x4c 0x38 0x37 0x30 0x20 0x20 0x20 0x20 0x20             
13: 0x0c 0x30 0x31 0x4b 0x4c 0x38 0x37 0x30 0x20 0x20 0x20 0x20 0x20             
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get power supply serial numbers:                             
# i2cget -f -y 3 0x69 0x9e i 7; i2cget -f -y 3 0x68 0x9e i 7     
7: 0x06 0x37 0x37 0x43 0x31 0x31 0x30                                            
7: 0x06 0x37 0x37 0x43 0x31 0x32 0x41                                            
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get power supply CCINs                                       
# i2cget -f -y 3 0x69 0xbd i 2; i2cget -f -y 3 0x68 0xbd i 2     
2: 0x2b 0x1d                                                                     
2: 0x2b 0x1d                                                                     
# #~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       
# # Get power supply software revision:                          
# i2cget -f -y 3 0x69 0xfa b; i2cget -f -y 3 0x69 0xfb b; i2cget -f -y 3 0x69 0xfc b; i2cget -f -y 3 0x69 0xfd b
0x01                                                                                                                            
0x12                                                                                                                            
0x01                                                                                                                            
0x14                                                                                                                            
# i2cget -f -y 3 0x68 0xfa b; i2cget -f -y 3 0x68 0xfb b; i2cget -f -y 3 0x68 0xfc b; i2cget -f -y 3 0x68 0xfd b
0x01                                                                                                                            
0x12                                                                                                                            
0x01                                                                                                                            
0x14
```

bjwyman, 2017-09-25:
```
# grep . ls /sys/kernel/debug/pmbus/hwmon3/*
grep: ls: No such file or directory                         
/sys/kernel/debug/pmbus/hwmon3/status0:0x0002               
/sys/kernel/debug/pmbus/hwmon3/status0_fan12:0x00           
/sys/kernel/debug/pmbus/hwmon3/status0_input:0x00           
/sys/kernel/debug/pmbus/hwmon3/status0_iout:0x00            
/sys/kernel/debug/pmbus/hwmon3/status0_temp:0x00            
/sys/kernel/debug/pmbus/hwmon3/status0_vout:0x00
```
```
# i2cget -f -y 3 0x68 0x79 w; i2cget -f -y 3 0x68 0x7D b; i2cget -f -y 3 0x68 0x80 b
0x0002                                                                                              
0x00                                                                                                
0x00
```

bjwyman, 2017-09-25:
```
# i2cget -f -y 3 0x68 0x7e b             
0x80
```

bjwyman, 2017-09-26:
Found another system with PN 01KL870, similar behavior.
Compared to system with older power supply, it looks like we end up missing status0_cml, status0_mfr, and status0_other.

amboar, 2017-10-04:
@bjwyman can you apply this patch and tell me whether you see different behaviour? http://patchwork.ozlabs.org/patch/809974/

bjwyman, 2017-10-04:
Doesn't appear to help get `status0_mfr` and `status0_other` to show up. CML bit is still on as well.
```
root@witherspoon:~# date; uptime
Wed Oct  4 23:16:24 UTC 2017
 23:16:24 up 18 min,  load average: 0.33, 0.35, 0.45
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-104"
VERSION_ID="v1.99.10-104-g794f874-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-104"
BUILD_ID="v1.99.10"
root@witherspoon:~# ls /sys/bus/i2c/devices/
11-0032  11-0060  3-0052   3-0068   3-0076   4-004c   4-0071   5-0070   9-004a   i2c-11   i2c-13   i2c-3    i2c-5
11-0051  11-0064  3-0060   3-0069   3-0077   4-0070   5-004c   5-0071   i2c-10   i2c-12   i2c-2    i2c-4    i2c-9
root@witherspoon:~# ls /sys/bus/i2c/devices/3-0068/hwmon/
hwmon3
root@witherspoon:~# ls /sys/bus/i2c/devices/3-0069/hwmon/
hwmon4
root@witherspoon:~# ls /sys/kernel/debug/pmbus/hwmon
hwmon10/ hwmon3/  hwmon4/  hwmon5/  hwmon6/  hwmon7/  hwmon8/  hwmon9/
root@witherspoon:~# ls /sys/kernel/debug/pmbus/hwmon3/
status0        status0_fan12  status0_input  status0_iout   status0_temp   status0_vout
root@witherspoon:~# ls /sys/kernel/debug/pmbus/hwmon4/
status0        status0_fan12  status0_input  status0_iout   status0_temp   status0_vout
root@witherspoon:~# grep . /sys/kernel/debug/pmbus/hwmon3/*
/sys/kernel/debug/pmbus/hwmon3/status0:0x1842
/sys/kernel/debug/pmbus/hwmon3/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_vout:0x00
root@witherspoon:~# grep . /sys/kernel/debug/pmbus/hwmon4/*
/sys/kernel/debug/pmbus/hwmon4/status0:0x1842
/sys/kernel/debug/pmbus/hwmon4/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_vout:0x00
root@witherspoon:~#
```

bjwyman, 2017-10-04:
Did a bitbake build with the patch in a file, and the obmc-ubi-fs removed from that one conf file, so that I could netboot the image(s).
```
(wspoonsdk)[v2cib530@gfwr703 linux-obmc]$ pwd                                          
/esw/san5/v2cib530/openbmc/openbmc/meta-phosphor/common/recipes-kernel/linux/linux-obmc
(wspoonsdk)[v2cib530@gfwr703 linux-obmc]$ ls *pmbus*                                   
pmbus-Fix-error-propagation-for-register-check-functions.patch                         
(wspoonsdk)[v2cib530@gfwr703 linux-obmc]$
```

bjwyman, 2017-10-04:
Files still missing after `obmcutil chassison`
```
root@witherspoon:~# ##########################
root@witherspoon:~# # Read top PSU STATUS_WORD
root@witherspoon:~# i2cget -f -y 3 0x69 0x79 w
0x0002                                        
root@witherspoon:~# # Read top PSU STATUS_INPUT
root@witherspoon:~# i2cget -f -y 3 0x69 0x7c b 
0x00                                           
root@witherspoon:~# # Read top PSU READ_VIN
root@witherspoon:~# i2cget -f -y 3 0x69 0x88 w
0xf9a0
root@witherspoon:~# #############################
root@witherspoon:~# # Read bottom PSU STATUS_WORD
root@witherspoon:~# i2cget -f -y 3 0x68 0x79 w
0x0002
root@witherspoon:~# # Read bottom PSU STATUS_INPUT
root@witherspoon:~# i2cget -f -y 3 0x68 0x7c b
0x00
root@witherspoon:~# # Read bottom PSU READ_VIN
root@witherspoon:~# i2cget -f -y 3 0x68 0x88 w
0xf99d
root@witherspoon:~# grep . /sys/kernel/debug/pmbus/hwmon3/*
/sys/kernel/debug/pmbus/hwmon3/status0:0x0002
/sys/kernel/debug/pmbus/hwmon3/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_vout:0x00
root@witherspoon:~# Oct 04 23:21:36 witherspoon avahi-daemon[927]: Service name conflict for "phosphor_rest" (/services/phosphor_rest.service), retrying with "phosphor_rest #35".
Oct 04 23:21:36 witherspoon avahi-daemon[927]: Service name conflict for "sftp" (/services/sftp.service), retrying with "sftp #35".

root@witherspoon:~# grep . /sys/kernel/debug/pmbus/hwmon4/*
/sys/kernel/debug/pmbus/hwmon4/status0:0x0002
/sys/kernel/debug/pmbus/hwmon4/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_vout:0x00
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# 
```

amboar, 2017-10-04:
The patch won't help so much as properly propagate an error if it occurs on either setting the PMBus page or doing a CLEAR_FAULTS operation. It might not save you if you're trying to access the status word and failing, as errors are ignored followed by a fall-back to probing the status byte register.

bjwyman, 2017-10-05:
Is there something else I should be looking at?

What sort of different behavior should I have been looking to notice?

amboar, 2017-10-09:
@bjwyman failing to probe, or frequent errors reported when reading sysfs attributes are some expected outcomes from that patch. Note there are some related patches to make sure the MAX31785 driver keeps working properly.

If you're keen to hack the kernel to resolve this I suggest you add some logging in the PMBus core to determine what failures are occurring.

amboar, 2018-04-16:
@bjwyman what's the state of this issue?

bjwyman, 2018-06-04:
As far as I know, this problem can still occur.

bjwyman, 2019-03-29:
Issue #2403 referenced this. That points to a commit that lessened the flood of messages when this occurs.
There was also a ibm-cffps kernel driver patch that I recall @eddiejames submitting to ignore the CML bit during the driver probing. I do not recall if that somehow avoided this. That was a change to address a VOUT_MODE problem.

bjwyman, 2019-03-29:
Issue #2727 is the one that has notes on the patch to skip the CML bit check.
Looks like that is merged into the kernel as: 4471879acf926221231fe68ebc30f9cad0d7f7c8.

bjwyman, 2019-03-29:
VERSION_ID="2.7.0-dev-233-g24b2063-dirty"
```
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/bus/i2c/devices/3-0068/hwmon/
hwmon3
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/bus/i2c/devices/3-0069/hwmon/
hwmon4
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/kernel/debug/pmbus/hwmon3/
cffps1         status0_cml    status0_input  status0_mfr    status0_temp
status0        status0_fan12  status0_iout   status0_other  status0_vout
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/kernel/debug/pmbus/hwmon4/
cffps1         status0_cml    status0_input  status0_mfr    status0_temp
status0        status0_fan12  status0_iout   status0_other  status0_vout
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# grep . /sys/kernel/debug/pmbus/hwmon3/*
/sys/kernel/debug/pmbus/hwmon3/status0:0x0842
/sys/kernel/debug/pmbus/hwmon3/status0_cml:0x80
/sys/kernel/debug/pmbus/hwmon3/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_mfr:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_other:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_vout:0x00
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# grep . /sys/kernel/debug/pmbus/hwmon4/*
/sys/kernel/debug/pmbus/hwmon4/status0:0x0842
/sys/kernel/debug/pmbus/hwmon4/status0_cml:0x80
/sys/kernel/debug/pmbus/hwmon4/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_mfr:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_other:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_vout:0x00
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~#
```
Build a patch to revert that CML bit checking:
```
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/bus/i2c/devices/3-0068/hwmon/
hwmon3
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/bus/i2c/devices/3-0069/hwmon/
hwmon4
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/kernel/debug/pmbus/hwmon3/
cffps1         status0        status0_fan12  status0_input  status0_iout   status0_temp   status0_vout
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# ls /sys/kernel/debug/pmbus/hwmon4/
cffps1         status0        status0_fan12  status0_input  status0_iout   status0_temp   status0_vout
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# grep . /sys/kernel/debug/pmbus/hwmon3/*
/sys/kernel/debug/pmbus/hwmon3/status0:0x0842
/sys/kernel/debug/pmbus/hwmon3/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon3/status0_vout:0x00
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# grep . /sys/kernel/debug/pmbus/hwmon4/*
/sys/kernel/debug/pmbus/hwmon4/status0:0x0842
/sys/kernel/debug/pmbus/hwmon4/status0_fan12:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_input:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_iout:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_temp:0x00
/sys/kernel/debug/pmbus/hwmon4/status0_vout:0x00
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-dev"
VERSION_ID="2.7.0-dev-234-gd9adce3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
BUILD_ID="2.7.0-dev"
OPENBMC_TARGET_MACHINE="witherspoon"
root@witherspoonccfba8cd1dc645aba2a3eeb0cdb39f76:~#
```
So, it was that patch that fixed this problem:
https://github.com/openbmc/linux/commit/4471879acf926221231fe68ebc30f9cad0d7f7c8#diff-225e4f85c83ed9ffae2e26e5e2e9f0c5

closed 2019-03-29 by bjwyman