openbmc/openbmc #2365: phosphor-mboxd: Flash read where there is no partition segfaults daemon

sjitindarsingh, 2017-09-26:
It is possible to perform a flash read on the host which will cause the daemon on the bmc side to segfault.

On the host:

/ # pflash --info
Flash info:
-----------
Name          = /dev/mtd0
Total size    = 64MB	 Flags E:ECC, P:PRESERVED, R:READONLY
Erase granule =  4KB           B:BACKUP, F:REPROVISION

TOC@0x00000000 Partitions:
-----------
ID=00            part 0x00001000..0x00002000 (actual=0x00001000) [-----]
ID=01            HBEL 0x00002000..0x00026000 (actual=0x00024000) [E----]
ID=02           GUARD 0x00026000..0x0002b000 (actual=0x00005000) [EP---]
ID=03           NVRAM 0x0002b000..0x000bb000 (actual=0x00090000) [-P---]
ID=04         SECBOOT 0x000bb000..0x000df000 (actual=0x00024000) [EP---]
ID=05           DJVPD 0x000df000..0x00127000 (actual=0x00048000) [E----]
ID=06            MVPD 0x00127000..0x001b7000 (actual=0x00090000) [E----]
ID=07            CVPD 0x001b7000..0x001ff000 (actual=0x00048000) [E----]
ID=08             HBB 0x001ff000..0x002ff000 (actual=0x00100000) [E----]
ID=09             HBD 0x002ff000..0x0041f000 (actual=0x00120000) [E----]
ID=10             HBI 0x0041f000..0x0107f000 (actual=0x00c60000) [E----]
ID=11             SBE 0x0107f000..0x0113b000 (actual=0x000bc000) [E----]
ID=12           HCODE 0x0113b000..0x0125b000 (actual=0x00120000) [E----]
ID=13            HBRT 0x0125b000..0x016db000 (actual=0x00480000) [E----]
ID=14         PAYLOAD 0x016db000..0x017db000 (actual=0x00100000) [-----]
ID=15      BOOTKERNEL 0x017db000..0x026db000 (actual=0x00f00000) [-----]
ID=16             OCC 0x026db000..0x027fb000 (actual=0x00120000) [E----]
ID=17         FIRDATA 0x027fb000..0x027fe000 (actual=0x00003000) [E----]
ID=18            CAPP 0x027fe000..0x02822000 (actual=0x00024000) [E----]
ID=19         BMC_INV 0x02822000..0x0282b000 (actual=0x00009000) [-----]
ID=20            HBBL 0x0282b000..0x02832000 (actual=0x00007000) [E----]
ID=21        ATTR_TMP 0x02832000..0x0283a000 (actual=0x00008000) [-----]
ID=22       ATTR_PERM 0x0283a000..0x02842000 (actual=0x00008000) [E----]
ID=23         VERSION 0x02842000..0x02843000 (actual=0x00001000) [-----]
ID=24     IMA_CATALOG 0x02843000..0x02883000 (actual=0x00040000) [E----]
ID=25         RINGOVD 0x02883000..0x028a3000 (actual=0x00020000) [-----]
ID=26         WOFDATA 0x028a3000..0x02ba3000 (actual=0x00300000) [E----]
ID=27     HB_VOLATILE 0x02ba3000..0x02ba8000 (actual=0x00005000) [E----]
ID=28            MEMD 0x02ba8000..0x02baf000 (actual=0x00007000) [E----]
ID=29            SBKT 0x02baf000..0x02bb3000 (actual=0x00004000) [E----]

As we can see there is nothing at 0x3000000. However this is still within the size of the flash.
Now try to read there:

/ # pflash -r /tmp/tmp -a 0x3000000 -s 0x100000
Reading to "/tmp/tmp" from 0x03000000..0x03100000 !
[                                                  ] 0%
[  447.050185421,3] MBOX-FLASH: BMC NOT RESPONDING 10 second wait
[  448.050254187,3] MBOX-FLASH: BMC NOT RESPONDING 11 second wait
[  449.050331435,3] MBOX-FLASH: BMC NOT RESPONDING 12 second wait
[  450.050404422,3] MBOX-FLASH: BMC NOT RESPONDING 13 second wait
[  451.050490557,3] MBOX-FLASH: BMC NOT RESPONDING 14 second wait
[  452.050572334,3] MBOX-FLASH: BMC NOT RESPONDING 15 second wait
[  453.050640660,3] MBOX-FLASH: BMC NOT RESPONDING 16 second wait
[  454.050706810,3] MBOX-FLASH: BMC NOT RESPONDING 17 second wait
[  455.050770889,3] MBOX-FLASH: BMC NOT RESPONDING 18 second wait
[  456.050837204,3] MBOX-FLASH: BMC NOT RESPONDING 19 second wait
[  457.050907814,3] MBOX-FLASH: BMC NOT RESPONDING 20 second wait
[  458.050983542,3] MBOX-FLASH: BMC NOT RESPONDING 21 second wait
[  459.051060691,3] MBOX-FLASH: BMC NOT RESPONDING 22 second wait
[  460.051137969,3] MBOX-FLASH: BMC NOT RESPONDING 23 second wait
[  461.051206166,3] MBOX-FLASH: BMC NOT RESPONDING 24 second wait
[  462.051276711,3] MBOX-FLASH: BMC NOT RESPONDING 25 second wait
[  463.051345394,3] MBOX-FLASH: BMC NOT RESPONDING 26 second wait
[  464.051418083,3] MBOX-FLASH: BMC NOT RESPONDING 27 second wait
[  465.051493166,3] MBOX-FLASH: BMC NOT RESPONDING 28 second wait
[  466.051570334,3] MBOX-FLASH: BMC NOT RESPONDING 29 second wait
[  467.051645403,3] MBOX-FLASH: BMC NOT RESPONDING 30 second wait
[  468.051722540,3] MBOX-FLASH: BMC NOT RESPONDING 31 second wait
[  468.154183441,3] MBOX-FLASH: Timeout waiting for BMC
[  468.154235149,3] MBOX-FLASH: Error waiting for BMC
Flash read error 15 for chunk at 0x03000000

And on the daemon (BMC) side:

[ 1506389173.798598983] MBOX Event
[ 1506389173.798795508] Received MBOX request:
[ 1506389173.798890336] MBOX cmd: 4
[ 1506389173.798971064] MBOX seq: 39
[ 1506389173.799050377] MBOX arg[0]: 0x00
[ 1506389173.799134862] MBOX arg[1]: 0x30
[ 1506389173.799211387] MBOX arg[2]: 0x00
[ 1506389173.799284761] MBOX arg[3]: 0x00
[ 1506389173.799362579] MBOX arg[4]: 0x00
[ 1506389173.799438175] MBOX arg[5]: 0x00
[ 1506389173.799513286] MBOX arg[6]: 0x00
[ 1506389173.799591953] MBOX arg[7]: 0x00
[ 1506389173.799666175] MBOX arg[8]: 0x00
[ 1506389173.799739225] MBOX arg[9]: 0x00
[ 1506389173.799814175] MBOX arg[10]: 0x00
[ 1506389173.799891306] Received MBOX command: 4
[ 1506389173.799971993] Close current window, flags: 0x00
[ 1506389173.800061569] Host requested flash @ 0x03000000
[ 1506389173.800143347] Searching for window which contains 0x03000000 
[ 1506389173.800238862] No existing window which maps that flash offset
[ 1506389173.800315104] Creating window which maps 0x03000000 
[ 1506389173.800403023] Searching for window which contains 0xffffffff exactly
[ 1506389173.800492114] No uninitialised window, evicting one
[ 1506389173.800564438] Copy virtual pnor to 0x74b82000 for size 0x00100000 from offset 0x03000000

[1]+  Segmentation fault      (core dumped) mboxd -f 64M -w 1M -vv

There is now no way for the host to access the flash unless you access the daemon and restart the bmc & reboot the host.

amboar, 2017-10-04:
@sjitindarsingh first up, a nit: Please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

Separately, to confirm, this is an issue with phosphor-mboxd built without `--enable-virtual-pnor`?

amboar, 2017-10-04:
> Separately, to confirm, this is an issue with phosphor-mboxd built without --enable-virtual-pnor?

Ah, I misread, you said at the top:

> perform a flash read on the host

ojayanth, 2017-10-25:
Recreated this issue and collected core.

#1  0x0002d0c4 in copy_flash (context=0xd7ef10, offset=50331648, mem=0x73562000, size=1048576) at mboxd_flash_virtual.cpp:151
151	            vpnor_get_partition_table_size(context) << context->block_size_shift;

amboar, 2017-10-25:
@ojayanth great. Do you have the bandwidth to fix it? Maybe @dkodihal has some thoughts.

ojayanth, 2017-10-27:
Identified the root cause for this issue , In const pnor_partition& partition(size_t offset)  function 
returns " reference to a zeroed out pnor_partition structure, incase fails to find partion for the given 
offset" . 
Copy_flash functions get incorrect partioninfo and evntally causing Segfault during ``` memcpy(mem, (char*)mapped_mem + diffOffset , rc);''`.

Proposed fix is Throw an exception incase if the requested offset is not within the partition limit.

ojayanth, 2017-10-27:
@amboar The proposed fix is to restrict the segfaults from mbox and return error back to Host.

What is the real requirement to enable support flash read, from offset outside partition ?

amboar, 2017-10-27:
The requirements are two fold:

1. The behaviour between virtualised and non-virtualised PNORs from the perspective of the host should be as similar as possible. Currently this is violated on a number of fronts, including this one. Anything otherwise needs to be documented and discussed with the hostboot and OPAL communities before being implemented and rolled out
2. The flash should not become inaccessible due to plausible access from the host. This is a denial of service vector. The host must not be able to crash processes on the BMC

ojayanth, 2017-10-27:
@amboar This issue will be used to fix segfault. Remaining points mentioned here will be tracked using new issue , which require more community level discussion.

ojayanth, 2017-10-27:
#2524 created to address @amboar comments.

rfrandse, 2017-10-31:
<https://gerrit.openbmc-project.xyz/7688> Fix for mboxd segfault during flash read
Resolves  #2365 phosphor-mboxd: Flash read where there is no partition segfaults daemon

closed 2017-11-17 by geissonator