openbmc/openbmc #2367: DC cycle test fail on zaius

ryandcyu, 2017-09-26:
@anoo1 @mine260309 
I do the stress testing for BMC, but it looks very unstable.
There have the kernel panic and can not power on the system during DC cycle.
Below are pnor and BMC version in our testing environment

Pnor:
```
op-build-v1.19-37-g7819c93
       buildroot-2017.08-5-g08caaab
       skiboot-v5.8-42-gbf1751cc6b4b
       hostboot-9e01538
       linux-4.13-openpower1-p123df2d
       petitboot-v1.5.1-pc20f80e
       machine-xml-fdb76ec
       occ-6f77ee7
       hostboot-binaries-67a15fd
       capp-ucode-9c73e9f
       sbe-8d90ab2
```
BMC:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-0"
VERSION_ID="v1.99.10-0-g357d010-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-0"
BUILD_ID="v1.99.10"
```
Some error message as below.
ex.1
```
Sep 20 16:07:11 zaius kernel: INFO: task openpower-occ-c:1088 blocked for more than 120 seconds.
Sep 20 16:07:11 zaius kernel:       Not tainted 4.10.17-c4227b43277bbd0992d10d764d4958239786503d #2
Sep 20 16:07:11 zaius kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 20 16:07:11 zaius kernel: openpower-occ-c D    0  1088      1 0x00000000
Sep 20 16:07:11 zaius kernel: (__schedule) from (schedule+0x54/0xac)
Sep 20 16:07:11 zaius kernel: (schedule) from (__kernfs_remove+0x1c4/0x260)
Sep 20 16:07:11 zaius kernel: (__kernfs_remove) from (kernfs_remove_by_name_ns+0x48/0x94)
Sep 20 16:07:11 zaius kernel: (kernfs_remove_by_name_ns) from (remove_files+0x48/0x94)
Sep 20 16:07:11 zaius kernel: (remove_files) from (sysfs_remove_group+0x50/0xa4)
Sep 20 16:07:11 zaius kernel: (sysfs_remove_group) from (sysfs_remove_groups+0x34/0x44)
Sep 20 16:07:11 zaius kernel: (sysfs_remove_groups) from (device_remove_attrs+0x4c/0x7c)
Sep 20 16:07:11 zaius kernel: (device_remove_attrs) from (device_del+0x1ec/0x314)
Sep 20 16:07:11 zaius kernel: (device_del) from (device_unregister+0x24/0x6c)
Sep 20 16:07:11 zaius kernel: (device_unregister) from (hwmon_device_unregister+0x44/0x84)
Sep 20 16:07:11 zaius kernel: (hwmon_device_unregister) from (devm_hwmon_release+0x1c/0x20)
Sep 20 16:07:11 zaius kernel: (devm_hwmon_release) from (release_nodes+0x164/0x1dc)
Sep 20 16:07:11 zaius kernel: (release_nodes) from (devres_release_all+0x34/0x54)
Sep 20 16:07:11 zaius kernel: (devres_release_all) from (device_release_driver_internal+0x154/0x1d8)
Sep 20 16:07:11 zaius kernel: (device_release_driver_internal) from (device_release_driver+0x20/0x24)
Sep 20 16:07:11 zaius kernel: (device_release_driver) from (unbind_store+0x84/0x104)
Sep 20 16:07:11 zaius kernel: (unbind_store) from (drv_attr_store+0x30/0x3c)
Sep 20 16:07:11 zaius kernel: (drv_attr_store) from (sysfs_kf_write+0x50/0x54)
Sep 20 16:07:11 zaius kernel: (sysfs_kf_write) from (kernfs_fop_write+0xe4/0x1e4)
Sep 20 16:07:11 zaius kernel: (kernfs_fop_write) from (__vfs_write+0x3c/0x128)
Sep 20 16:07:11 zaius kernel: (__vfs_write) from (vfs_write+0xb0/0x178)
Sep 20 16:07:11 zaius kernel: (vfs_write) from (SyS_write+0x4c/0xa0)
Sep 20 16:07:11 zaius kernel: (SyS_write) from (ret_fast_syscall+0x0/0x34)
```
ex.2
```
[ 2305.510000] BUG: Bad page map in process phosphor-fan-co  pte:bd42e400 pmd:00000000
[ 2305.550000] addr:44200000 vm_flags:00100071 anon_vma:bd67a9a0 mapping:bd9519dc index:442a0
[ 2305.550000] file:libm-2.24.sofault:filemap_fault mmap:generic_file_readonly_mmap readpage:squashfs_readpage
[ 2305.550000] CPU: 0 PID: 7386 Comm: phosphor-fan-co Tainted: G    B D W       4.10.17-6cf936f754c600944e9b046245c3cc592e07c482 #1
[ 2305.550000] Hardware name: ASpeed SoC
[ 2305.550000] (unwind_backtrace) from (show_stack+0x20/0x24)
[ 2305.550000] (show_stack) from (dump_stack+0x20/0x28)
[ 2305.600000] (dump_stack) from (print_bad_pte+0x130/0x1a8)
[ 2305.610000] (print_bad_pte) from (unmap_page_range+0x11c/0x52c)
[ 2305.610000] (unmap_page_range) from (unmap_single_vma+0x50/0x58)
[ 2305.620000] (unmap_single_vma) from (unmap_vmas+0x64/0x78)
[ 2305.630000] (unmap_vmas) from (exit_mmap+0xe8/0x1f8)
[ 2305.630000] (exit_mmap) from (mmput+0x40/0xc4)
[ 2305.650000] (mmput) from (do_exit+0x230/0x8e8)
[ 2305.650000] (do_exit) from (do_group_exit+0x4c/0xb4)
[ 2305.650000] (do_group_exit) from (get_signal+0x214/0x580)
[ 2305.670000] (get_signal) from (do_signal+0xd4/0x3dc)
[ 2305.670000] (do_signal) from (do_work_pending+0xac/0xbc)
[ 2305.670000] (do_work_pending) from (slow_work_pending+0xc/0x20)
[ 2305.670000] BUG: Bad page map in process phosphor-fan-co  pte:bd70f400 pmd:00000000
[ 2305.700000] addr:44201000 vm_flags:00100071 anon_vma:bd67a9a0 mapping:bd9519dc index:442a1
[ 2305.700000] file:libm-2.24.sofault:filemap_fault mmap:generic_file_readonly_mmap readpage:squashfs_readpage 
```
ex.3
```
[  307.350000] BUG: Bad page map in process phosphor-hwmon-  pte:7f000000 pmd:00000000
[  307.350000] addr:00c02203 vm_flags:00000875 anon_vma:  (null) mapping:bd0e73fc index:bf2
[  307.350000] file:phosphor-hwmon-readdfault:filemap_fault mmap:generic_file_readonly_mmap readpage:squashfs_readpage
[  307.350000] CPU: 0 PID: 1786 Comm: phosphor-hwmon- Not tainted 4.10.17-6cf936f754c600944e9b046245c3cc592e07c482 #1
[  307.350000] Hardware name: ASpeed SoC
[  307.350000] (unwind_backtrace) from (show_stack+0x20/0x24)
[  307.350000] (show_stack) from (dump_stack+0x20/0x28)
[  307.350000] (dump_stack) from (print_bad_pte+0x130/0x1a8)
[  307.350000] (print_bad_pte) from (unmap_page_range+0x11c/0x52c)
[  307.350000] (unmap_page_range) from (unmap_single_vma+0x50/0x58)
[  307.350000] (unmap_single_vma) from (unmap_vmas+0x64/0x78)
[  307.350000] (unmap_vmas) from (exit_mmap+0xe8/0x1f8)
[  307.350000] (exit_mmap) from (mmput+0x40/0xc4)
[  307.350000] (mmput) from (do_exit+0x230/0x8e8)
[  307.350000] (do_exit) from (do_group_exit+0x4c/0xb4)
[  307.350000] (do_group_exit) from (SyS_exit_group+0x20/0x24)
[  307.350000] (SyS_exit_group) from (ret_fast_syscall+0x0/0x34)
[  307.350000] Disabling lock debugging due to kernel taint
```
Then, I remove the "openpower-occ-control" under meta-openbmc-machines/meta-openpower/meta-ingrasys/meta-zaius/recipes-phosphor/packagegroups
```
--- a/meta-openbmc-machines/meta-openpower/meta-ingrasys/meta-zaius/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend
+++ b/meta-openbmc-machines/meta-openpower/meta-ingrasys/meta-zaius/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend
@@ -1,2 +1,2 @@
 # Support OCC pass through and general occ control
-RDEPENDS_${PN}-inventory += "openpower-occ-control"
+# RDEPENDS_${PN}-inventory += "openpower-occ-control"
```
After remove "openpower-occ-control", I did not see the fail symptom about 75 times stress testing.
Anyone have any suggestions?

amboar, 2017-09-26:
This looks like a duplicate of #2345

geissonator, 2017-09-26:
Feel free to note as a dup and we can close it out and track in #2345 @eddiejames   We should get some more details in #2345 though.

geissonator, 2017-10-03:
Dup of #2345

closed 2017-10-03 by geissonator