openbmc/openbmc #2368: OP910:OPRAS:Witherspoon:Procedure callouts don't show up in BMC error logs

cdeadmin, 2017-09-26:
<cde:info> SW399603 </cde:info>

spinler, 2017-09-26:
Please investigate what the code is doing with the eSELs hostboot is sending down that just contain their procedure callout IDs.    We need to still create the eSEL style error logs for these.

vishwabmc, 2017-10-03:
I will start working on this now.

vishwabmc, 2017-10-03:
@dcrowell Is there a way to create eSEL at will and send it down ?. Looking at the defect, it seem to me that we may not want to interpret the info but continue posting as we have been posting.

But if something is being thrown, then need to investigate and hence need a way to create eSEL at will.

vishwabmc, 2017-10-12:
Couple things from the discussion.
```
1/. HB sends DF  ( basically a PEL ) down to BMC.
2/. If the PEL is supposed to have a high priority PROCEDURE CALLOUT, 
      then HB sends `DE` OEM-SEL 
       --> 4th byte is supposed to tell it's `DE`, and then the `Last 2 bytes in 16` should tell
             the ID of DF where this is associated with.
3/. If the PEL needs to have Hardware CALLOUTS, then HB will send  IPMI SELs
```

BMC seem to be not handling `DE` and hence the procedure callout info is getting lost. 

A quick look at `ipmi_ret_t ipmi_ibm_oem_partial_esel(ipmi_netfn_t netfn, ipmi_cmd_t cmd,` did not
show any switch statements and all went by the offset, so may be DE is not coming under this command func at all ?

But more important is that how we are interpreting DF and the SELs and how we need to interpret
the DE. I was suggesting that we don't interpret the DE and just post an OpenBMC error-log with those
16 bytes in it and make it a Host_Event and let the parser ( someone needs to write ) fetch these info
and make a meaning out of it.

rfrandse, 2017-10-31:
<https://gerrit.openbmc-project.xyz/7692> Commit a MaintenanceProcedure log entry on a 0xDE SEL record
Resolves  #2368 OP910:OPRAS:Witherspoon:Procedure callouts don't show up in BMC error logs

closed 2017-11-06 by geissonator