openbmc/openbmc #2372: BMC gets stuck in reboot loop after deleting activated image

charleshofer, 2017-09-26:
Steps to replicate

1. Started with -r5
2. Code update to -r7 and rebooted so that it was running
3. Uploaded and activated -r2 (do not reboot the BMC)
4. Deleted -r2
5. Rebooted

BMC will get stuck rebooting itself, giving the following error.

```
 4.600000] ubi4: max/mean erase counter: 7/4, WL threshold: 4096, image sequence number: 1358235831
[    4.600000] ubi4: available PEBs: 433, total reserved PEBs: 71, PEBs reserved for bad PEB handling: 0
[    4.600000] ubi4 error: ubi_open_volume: cannot open device 4, volume 0, error -19
[    4.600000] UBI: block: can't open volume on ubi4_0, err=-19
[    4.600000] console [netcon0] enabled
[    4.600000] netconsole: network logging started
[    4.600000] input: gpio-keys as /devices/platform/gpio-keys/input/input2
[    4.600000] ubi4: background thread "ubi_bgt4d" started, PID 506
[    4.660000] rtc-rv8803 11-0032: setting system clock to 2016-04-25 15:20:52 UTC (1461597652)
[    4.670000] VFS: Cannot open root device "ubiblock4_0" or unknown-block(0,0): error -6
[    4.670000] Please append a correct "root=" boot option; here are the available partitions:
```

saqibkh, 2017-09-26:
I think we can create a function inside item_updater that updates the u-boot environment variables after every version delete and after every priority change.

rfrandse, 2017-10-04:
<https://gerrit.openbmc-project.xyz/7067> Update Uboot Environement Variables when a version is delete or updated
Resolves  #2372 BMC gets stuck in reboot loop after deleting activated image

closed 2017-10-23 by geissonator