openbmc/openbmc #2374: phosphor-ipmi-host.service: cored dump during powering on

gkeishin, 2017-09-26:
This is pervasive on every power on but does boot to OS good so far. 

```
Sep 19 20:48:40 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=6/ABRT
Sep 19 20:48:40 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Sep 19 20:48:40 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Sep 19 20:48:42 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-42"
VERSION_ID="v1.99.10-42-ga101c6a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-42"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00cd954c889746a65f9022088c5d83f9f43bfddc #1 Tue Sep 26 18:05:33 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-8ee5291f/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.19_1.32
        op-build-e87d6b0
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-73-g4a2b8317fd3f
        hostboot-3d6c541
        linux-4.13-openpower1-pbe1de24
        petitboot-v1.5.1-p810ff34
        machine-xml-f8aaa73-pf7fdf7e
        occ-a43395b
        hostboot-binaries-836385d
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~#
```
- [x] Journald 
[IPMID_CORE_IPL.txt](https://github.com/openbmc/openbmc-test-automation/files/1334777/IPMID_CORE_IPL.txt)

gkeishin, 2017-09-27:
persistent in newer build too
```
Sep 19 20:51:11 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=6/ABRT
Sep 19 20:51:11 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Sep 19 20:51:11 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Sep 19 20:51:13 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
```
`Driver Info`

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-56"
VERSION_ID="v1.99.10-56-g5759007"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-56"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00cd954c889746a65f9022088c5d83f9f43bfddc #1 Fri Sep 22 12:34:59 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

tomjoseph83, 2017-09-27:
```
Sep 19 20:52:26 witherspoon ipmid[1782]: Bus: :1.5, Path: /xyz/openbmc_project/control/host0/TPMEnable, Interface: xyz.openbmc_project.Control.TPM.Policy
Sep 19 20:52:33 witherspoon ipmid[1782]: ERROR:[0x80] handling NetFn:[0x2C], Cmd:[0x3]
Sep 19 20:52:33 witherspoon ipmid[1782]: Failed to get /xyz/openbmc_project/control/host0/powersupply_derating busname: (null)
Sep 19 20:52:33 witherspoon ipmid[1782]: Failed to find Sensor 0xd8
Sep 19 20:52:33 witherspoon ipmid[1782]: terminate called after throwing an instance of 'std::runtime_error'
```

This probably happened because issue no #2154 was partially merged.

rfrandse, 2017-09-28:
<https://gerrit.openbmc-project.xyz/6977> Handle std::runtime_error exception in get sensor reading
Resolves  #2374 phosphor-ipmi-host.service: cored dump during powering on

closed 2017-10-03 by geissonator