openbmc/openbmc #2377: phosphor-inventory: Encountered unsupported interface

gkeishin, 2017-09-27:
```
root@witherspoon:~# journalctl --no-pager | grep phosphor-inventory
Sep 19 20:40:05 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:25 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:25 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:26 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
Sep 19 20:51:27 witherspoon phosphor-inventory[1037]: Encountered unsupported interface.
root@witherspoon:~#
```

`JSON pretty print`
```
{
	"__CURSOR" : "s=03a0ca9462e94d92afac880cd8ee63ba;i=2d7;b=e8c7108b719f4a03aa45879ce92a7480;m=66acfef;t=55990dd4298f5;x=f5f0bb0bee664a2b",
	"__REALTIME_TIMESTAMP" : "1505853605845237",
	"__MONOTONIC_TIMESTAMP" : "107663343",
	"_BOOT_ID" : "e8c7108b719f4a03aa45879ce92a7480",
	"_MACHINE_ID" : "3428902596a04e44a1f883c6daaf1787",
	"_HOSTNAME" : "witherspoon",
	"_UID" : "0",
	"_GID" : "0",
	"_SYSTEMD_SLICE" : "system.slice",
	"_TRANSPORT" : "journal",
	"_CAP_EFFECTIVE" : "3fffffffff",
	"PRIORITY" : "3",
	"CODE_FILE" : "/usr/include/phosphor-logging/log.hpp",
	"CODE_LINE" : "100",
	"CODE_FUNC" : "helper_log",
	"MESSAGE" : "Encountered unsupported interface.",
	"TRANSACTION_ID" : "607442687",
	"INTERFACE" : [ 8, 9, 124, 1, 43 ],
	"SYSLOG_IDENTIFIER" : "phosphor-inventory",
	"_PID" : "1037",
	"_COMM" : "phosphor-invent",
	"_EXE" : "/usr/sbin/phosphor-inventory",
	"_CMDLINE" : "phosphor-inventory",
	"_SYSTEMD_CGROUP" : "/system.slice/xyz.openbmc_project.Inventory.Manager.service",
	"_SYSTEMD_UNIT" : "xyz.openbmc_project.Inventory.Manager.service",
	"_SYSTEMD_INVOCATION_ID" : "5fd3bdf9146944a39cd463011b14055d",
	"_SOURCE_REALTIME_TIMESTAMP" : "1505853605842786"
}
```
`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-56"
VERSION_ID="v1.99.10-56-g5759007"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-56"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00cd954c889746a65f9022088c5d83f9f43bfddc #1 Fri Sep 22 12:34:59 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-27b41bda/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.19_1.41
        op-build-v1.7-1383-g487ffd1
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-90-g398a98123641-pb7fe7a5
        hostboot-3d6c541
        linux-4.13-openpower1-p1b975d8
        petitboot-v1.5.1-p65d605b
        machine-xml-f8aaa73-p03357dc
        occ-a43395b
        hostboot-binaries-edd568f
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~#
```

- [x] Journald 
[Inventory_error.txt](https://github.com/openbmc/openbmc/files/1337406/Inventory_error.txt)

navrathi, 2017-11-02:
@gkeishin As discussed this issue is not reproducible with the latest code update. Recommend closing it.

gkeishin, 2017-11-02:
I have not seen this in latter build.. Not sure which track fixed it..

dkodihal, 2017-11-30:
Please reopen if this is seen again.

closed 2017-11-30 by dkodihal