openbmc/openbmc #2378: Remove org.openbmc.* from REST server

geissonator, 2017-09-27:


ngorugan, 2017-10-06:
- [ ] Create new map(container) with cap size of 10(configurable) for info logs

- [ ] Once we reach 10 info logs, send a warning/info msg and Clear oldest info log to make room for new  and store it.

- [ ] Use the existing map(container) with existing cap size of 100 to store error logs.

- [ ] Once we reach cap size, send a warning

dkodihal, 2017-10-10:
@ngorugan @bradbishop I'm thinking this can be a plug-in which runs on each request, checks if the request url consists of /org/openbmc, if so, fails with a 'not allowed'.

rfrandse, 2017-10-18:
<https://gerrit.openbmc-project.xyz/7422> Remove org.openbmc.* from REST server
Resolves  #2378 Remove org.openbmc.* from REST server

mine260309, 2018-01-26:
This breaks the code update for non-UBIFS BMCs, a.k.s all OpenBMC machines except Witherspoon.
I suggest revert it ASAP and commit it back until we have moved all services under `/org/openbmc/` to `/xyz/openbmc_project/`

geissonator, 2018-01-26:
We'll need to put this new code in it's own file, so it can easily be pulled in for just systems that want it.

rfrandse, 2018-04-18:
<https://gerrit.openbmc-project.xyz/10209> Remove org.openbmc.* from REST server
Resolves  #2378 Remove org.openbmc.* from REST server

closed 2018-07-11 by geissonator