openbmc/openbmc #2379: EnergyScale: Power Limit changes while deactivated

ssombar, 2017-09-27:
System: Witherspoon 

Test Case:
----------
Change Active Power Limit w/System Power ON and Deactivate Power Limit prior to change

Summary:
--------
Check the power limit is active and note the value, then deactivate the power limit.
Verify power limit is deactivated and set power limit to change current power limit.

Expected Results:
-----------------
Verify the changed power limit does not take effect.

Note: Need someone to review the results and verify if the Power Limit should not
      change while Power Limit is Deactivated.

Test Procedure:
---------------
Step 1: System at Runtime
Step 2: Get_Limit / Power Limit Active / Power Limit set to 500 Watts

        Current Limit State: Power Limit Active
        Exception actions:   Hard Power Off & Log Event to SEL
        Power Limit:         500 Watts
        Correction time:     0 milliseconds
        Sampling period:     0 seconds

Step 3: Deactivate Power Limit

        Power limit successfully deactivated

Step 4: Set Limit to 800 Watts (Note: Power Limit Deactivated, new Limit should NOT be set)

        Current Limit State: No Active Power Limit  <-------------Deactivated
        Exception actions:   Hard Power Off & Log Event to SEL
        Power Limit:         800 Watts              <-------------New Limit was accepted
        Correction time:     0 milliseconds
        Sampling period:     0 seconds

gkeishin, 2017-09-28:
@tomjoseph83 ^^^

tomjoseph83, 2017-10-03:
@ssombar Limit is a setting that works independently of whether the power limit is active. So the behaviour is as expected.

ssombar, 2017-10-03:
Interesting...since this was not the case for Firestone and Habanero.