openbmc/openbmc #2380: Handle new eSEL from host which will contain OCC telemetry data

geissonator, 2017-09-27:
The OCC generates a log every 24 hours with telemetry data (power, temperatures, ...).  The host will send this error down to the BMC when it gets it from the OCC. 

The BMC needs to do the following
- Handle the new eSEL OEM command (talk with Venkatesh/Dan C. for details)
- Log a new error, which will be informational, and have a custom Message for this

This design should piggy back of how BMC firmware handles the eSEL and the host callouts currently in host-ipmid.

tomjoseph83, 2017-10-05:
The Hostboot would be sending OCC metrics data as ESEL with the record type DD. The host currently sends ESEL with record type DF. The ESEL is send to BMC using the open power OEM command `Partial Add eSEL Data`, followed by `Add SEL` command which creates the logging entry. In the case of OCC metrics data, there is no sensor number associated with it, so Hostboot prefers that `Add SEL` command is not sent to create the logging entry.

- [ ] The progress indicator in the `Partial Add eSEL Data`  indicates whether the data received is the last chunk, in which case the command implementation is modified to check whether the record type is DD. If it is DD create a logging entry with the OCC metrics data.

- [ ]  Create a new logging yaml to describe the metrics data, with the severity as informational. It would be in the org/open_power/Host namespace.

rfrandse, 2017-10-17:
<https://gerrit.openbmc-project.xyz/7412> Copy the error yaml's for logging manager parsing
Resolves  #2380 Handle new eSEL from host which will contain OCC telemetry data

closed 2017-10-26 by geissonator