openbmc/openbmc #2382: STC:OP910:Witherspoon:OS reboot request leaves system in powered off state

cdeadmin, 2017-09-27:


cbostic, 2017-09-28:
```==== State: Open by: tdylla on 26 September 2017 12:35:14 ==== 1

System - yl13ub

BMC - yl13ubbmc.rch.stglabs.ibm.com (9.5.166.120)   root/0penBmc
OS  - yl13ubos.rch.stglabs.ibm.com  (9.5.166.121)   root/C0vfefed or root/Why4askme
OS Console Connect Instrutions:
1. ssh to bmc port 2200
2. signon with bmc userid & password
3. press enter and you should now be on OS console.  If system is at mon you can use ? to get control and help.

Firmware Code Levels
====================
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-redbud-ibm-OP9_v1.19_1.33
        op-build-450d9d7
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-73-g4a2b8317fd3f-p6faaa2d
        hostboot-3d6c541
        linux-4.13-openpower1-pe8fb223
        petitboot-v1.5.1-p36336db
        machine-xml-f8aaa73-p5482dba
        occ-a43395b
        hostboot-binaries-836385d
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"

BMC State
=========
root@witherspoon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

Problem Description:
====================
I removed A/C power from the system and then reapplied A/C power and powered the system on from the BMC with the obmcutil poweron command.  The system to the OS.  After running htx for a short time, the system was rebooted.  The system appeared to terminate normally in the os but never powered back up.  I check the BMC state of the power and States they appear to be in conflict with each other.

System is sitting in the failed state and being held.  Please connect to system and collect the debug information you need to correct the issue.  BMC dmesg output is below.

BMC DMESG OUTPUT
================

root@witherspoon:~# dmesg
[ 4544.650000] [<8023f728>] (device_del) from [<8023f86c>] (device_unregister+0x24/0x6c)
[ 4544.650000] [<8023f86c>] (device_unregister) from [<802cd308>] (__fsi_slave_remove_device+0x18/0x20)
[ 4544.710000] [<802cd308>] (__fsi_slave_remove_device) from [<8023d98c>] (device_for_each_child+0x64/0x90)
[ 4544.710000] [<8023d98c>] (device_for_each_child) from [<802cd2dc>] (__fsi_master_remove_slave+0x24/0x38)
[ 4544.710000] [<802cd2dc>] (__fsi_master_remove_slave) from [<8023d98c>] (device_for_each_child+0x64/0x90)
[ 4544.710000] [<8023d98c>] (device_for_each_child) from [<802ce03c>] (master_rescan_store+0x28/0x40)
[ 4544.710000] [<802ce03c>] (master_rescan_store) from [<8023c310>] (dev_attr_store+0x28/0x34)
[ 4544.710000] [<8023c310>] (dev_attr_store) from [<8014541c>] (sysfs_kf_write+0x50/0x54)
[ 4544.710000] [<8014541c>] (sysfs_kf_write) from [<80144b14>] (kernfs_fop_write+0xe4/0x1e4)
[ 4544.710000] [<80144b14>] (kernfs_fop_write) from [<800e24b4>] (__vfs_write+0x3c/0x128)
[ 4544.780000] [<800e24b4>] (__vfs_write) from [<800e2730>] (vfs_write+0xb0/0x178)
[ 4544.780000] [<800e2730>] (vfs_write) from [<800e28e4>] (SyS_write+0x4c/0xa0)
[ 4544.790000] [<800e28e4>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 4667.360000] INFO: task sh:2036 blocked for more than 120 seconds.
[ 4667.360000]       Tainted: G        W       4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1
[... 175 lines cut ...]
I opened SW403343 this morning which looks like it may in part be a recreate of the reboot issue in this defect.  My system yl13u8.aus.stglabs.ibm.com checkstopped, which initiated a reboot that did not succeed.  Now the system is showing

root@witherspoon:~$ obmcutil power
pgood = 1
state = 1
pgood_timeout = 10

root@witherspoon:~$ obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

and will not respond to poweron.  Currently holding in failed state.

==== State: Open by: rfrandse on 27 September 2017 16:03:42 ==== 8

#=#=# 2017-09-27 16:03:39 (CDT) #=#=#
New Component_Name = [openbmc_ext]
New OwnerInfo = [Bostic, Christopher l. (cbostic@us.ibm.com)]
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: cde00 on 27 September 2017 16:08:05 ==== 9

<cde:info>The defect was bridged to https://github.com/openbmc/openbmc/issues/2382 </cde:info>
```

cbostic, 2017-09-28:
#STATUS

I've rejected SW403240 as a duplicate of SW401584.  Expecting the cde bridge to close this issue out now.

closed 2017-09-28 by rfrandse