openbmc/openbmc #2383: STC:OP910:Witherspoon:OS reboot request leaves system in powered off state

cdeadmin, 2017-09-27:


cbostic, 2017-09-28:
SW403064 - Defect View - BestQuest | v 0.80.80 What's New? \| FAQProblem Report / Suggestion Request | v 0.80.80 What's New? \| FAQProblem Report / Suggestion Request
-- | -- | --
v 0.80.80 What's New? \| FAQProblem Report / Suggestion Request

cbostic, 2017-09-28:
System - yl13ub

BMC - yl13ubbmc.rch.stglabs.ibm.com (9.5.166.120)   root/0penBmc
OS  - yl13ubos.rch.stglabs.ibm.com  (9.5.166.121)   root/C0vfefed or root/Why4askme
OS Console Connect Instrutions:
 1. ssh to bmc port 2200
 2. signon with bmc userid & password
 3. press enter and you should now be on OS console.  If system is at mon you can use ? to get control and help.

Firmware Code Levels
====================
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-redbud-ibm-OP9_v1.19_1.33
        op-build-450d9d7
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-73-g4a2b8317fd3f-p6faaa2d
        hostboot-3d6c541
        linux-4.13-openpower1-pe8fb223
        petitboot-v1.5.1-p36336db
        machine-xml-f8aaa73-p5482dba
        occ-a43395b
        hostboot-binaries-836385d
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"

BMC State
=========
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

Problem Description:
====================
Running HTX soft bootme's.  When its time to reboot system, system shuts down okay but does not come back up.  System appears to be powered off when checking the system state on BMC.

System is sitting in the failed state and being held.  Please connect to system and collect the debug information you need to correct the issue.

==== State: Open by: cde00 on 25 September 2017 07:55:17 ==== 2

==== State: Open by: andrewg on 25 September 2017 10:51:47 ==== 3

Sep 25 06:31:20 witherspoon systemd[1]: Reached target Power0 On.
Sep 25 06:31:20 witherspoon systemd[1]: Reached target Power0 (On).
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Control Initialization.
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Monitor Initialization.
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Sep 25 06:31:20 witherspoon systemd[1]: Starting Scan FSI devices...
Sep 25 06:31:20 witherspoon systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Sep 25 06:31:20 witherspoon systemd[1]: Started Witherspoon Power Sequencer Power-on Monitor.
Sep 25 06:31:21 witherspoon kernel: Unable to handle kernel paging request at virtual address b00a8c92
Sep 25 06:31:21 witherspoon kernel: pgd = 8efe8000
Sep 25 06:31:21 witherspoon kernel: [b00a8c92] *pgd=00000000
Sep 25 06:31:21 witherspoon kernel: Internal error: Oops: 5 [#5] ARM
Sep 25 06:31:21 witherspoon kernel: CPU: 0 PID: 21060 Comm: sh Tainted: G    B D         4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1
Sep 25 06:31:21 witherspoon kernel: Hardware name: ASpeed SoC
Sep 25 06:31:21 witherspoon kernel: task: 9e6d8040 task.stack: 90cbc000
Sep 25 06:31:21 witherspoon kernel: PC is at strlen+0xc/0x38
Sep 25 06:31:21 witherspoon kernel: LR is at kobject_get_path+0x28/0x100
[... 90 lines cut ...]

Not sure if another flavor of https://w3.rchland.ibm.com/projects/bestquest/?defect=SW401584 or something different here.
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: andrewg on 25 September 2017 11:06:04 ==== 5

#=#=# 2017-09-25 11:06:03 (CDT) #=#=#
New OwnerInfo = [Bostic, Christopher l. (cbostic@us.ibm.com)]

Seems different so Chris is going to help out.  Eddie thought he had a fix for this in already so first thing to do is track down that change and verify it's in the .10-r7 build.
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: cbostic on 25 September 2017 15:49:30 ==== 6

Appears this build has the latest remove() fixes for i2c-fsi.c

commit e27d476ac884d4f4190456111425a83e58ee8083
Author: Edward A. James <eajames@us.ibm.com>
Date:   Mon Aug 7 22:01:42 2017 -0500

    i2c: fsi: Fix bug on remove() and reset to upstream state

This is similar to the other adapter remove issues but is slightly different so likely not a duplicate of those.

v 0.80.80 What's New? \| FAQProblem Report / Suggestion Request

cbostic, 2017-09-28:
```System - yl13ub

BMC - yl13ubbmc.rch.stglabs.ibm.com (9.5.166.120)   root/0penBmc
OS  - yl13ubos.rch.stglabs.ibm.com  (9.5.166.121)   root/C0vfefed or root/Why4askme
OS Console Connect Instrutions:
 1. ssh to bmc port 2200
 2. signon with bmc userid & password
 3. press enter and you should now be on OS console.  If system is at mon you can use ? to get control and help.

Firmware Code Levels
====================
root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-redbud-ibm-OP9_v1.19_1.33
        op-build-450d9d7
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-73-g4a2b8317fd3f-p6faaa2d
        hostboot-3d6c541
        linux-4.13-openpower1-pe8fb223
        petitboot-v1.5.1-p36336db
        machine-xml-f8aaa73-p5482dba
        occ-a43395b
        hostboot-binaries-836385d
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r7-0-gcce0f1b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r7"

BMC State
=========
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

Problem Description:
====================
Running HTX soft bootme's.  When its time to reboot system, system shuts down okay but does not come back up.  System appears to be powered off when checking the system state on BMC.

System is sitting in the failed state and being held.  Please connect to system and collect the debug information you need to correct the issue.

==== State: Open by: cde00 on 25 September 2017 07:55:17 ==== 2

==== State: Open by: andrewg on 25 September 2017 10:51:47 ==== 3

Sep 25 06:31:20 witherspoon systemd[1]: Reached target Power0 On.
Sep 25 06:31:20 witherspoon systemd[1]: Reached target Power0 (On).
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Control Initialization.
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Monitor Initialization.
Sep 25 06:31:20 witherspoon systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Sep 25 06:31:20 witherspoon systemd[1]: Starting Scan FSI devices...
Sep 25 06:31:20 witherspoon systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Sep 25 06:31:20 witherspoon systemd[1]: Started Witherspoon Power Sequencer Power-on Monitor.
Sep 25 06:31:21 witherspoon kernel: Unable to handle kernel paging request at virtual address b00a8c92
Sep 25 06:31:21 witherspoon kernel: pgd = 8efe8000
Sep 25 06:31:21 witherspoon kernel: [b00a8c92] *pgd=00000000
Sep 25 06:31:21 witherspoon kernel: Internal error: Oops: 5 [#5] ARM
Sep 25 06:31:21 witherspoon kernel: CPU: 0 PID: 21060 Comm: sh Tainted: G    B D         4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1
Sep 25 06:31:21 witherspoon kernel: Hardware name: ASpeed SoC
Sep 25 06:31:21 witherspoon kernel: task: 9e6d8040 task.stack: 90cbc000
Sep 25 06:31:21 witherspoon kernel: PC is at strlen+0xc/0x38
Sep 25 06:31:21 witherspoon kernel: LR is at kobject_get_path+0x28/0x100
[... 90 lines cut ...]

Not sure if another flavor of https://w3.rchland.ibm.com/projects/bestquest/?defect=SW401584 or something different here.
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: andrewg on 25 September 2017 11:06:04 ==== 5

#=#=# 2017-09-25 11:06:03 (CDT) #=#=#
New OwnerInfo = [Bostic, Christopher l. (cbostic@us.ibm.com)]

Seems different so Chris is going to help out.  Eddie thought he had a fix for this in already so first thing to do is track down that change and verify it's in the .10-r7 build.
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: cbostic on 25 September 2017 15:49:30 ==== 6

Appears this build has the latest remove() fixes for i2c-fsi.c

commit e27d476ac884d4f4190456111425a83e58ee8083
Author: Edward A. James <eajames@us.ibm.com>
Date:   Mon Aug 7 22:01:42 2017 -0500

    i2c: fsi: Fix bug on remove() and reset to upstream state

This is similar to the other adapter remove issues but is slightly different so likely not a duplicate of those.

```

cbostic, 2017-09-28:
STATUS

9/28/2017 3:30 PM

Continuing debug.  Not yet received any feedback from LTC team - will set up a meeting tonight to discuss further.    We fail when a kobject name string is found to not be a valid pointer - seemingly memory corruption.  This is based on the first set of data provided.

This issue is likely either a duplicate of SW401584 or a similar memory corruption issue as in issue 2388.   Latest data posted to the CQ for this issue indicates a new fail with 10-r9 version with fixes from Eddie but its debatable if the new fail mode is same issue as before.

cbostic, 2017-09-28:
Latest data from today with 10-r9 build:

```Just reproduced this problem on the same machine using a new BMC code level that Uma Yadlapati gave me.

root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10

root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

root@witherspoon:~# dmesg
[    2.100000] at24 11-0051: 8192 byte 24c64 EEPROM, writable, 1 bytes/write
[    2.100000] aspeed-i2c-bus 1e78a400.i2c-bus: i2c bus 11 registered, irq 32
[    2.100000] aspeed-i2c-bus 1e78a440.i2c-bus: i2c bus 12 registered, irq 33
[    2.100000] aspeed-i2c-bus 1e78a480.i2c-bus: i2c bus 13 registered, irq 34
[    2.100000] Driver for 1-wire Dallas network protocol.
[    2.100000] lm75 9-004a: hwmon0: sensor 'tmp275'
[    3.780000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[    4.480000] leds-pca955x 3-0060: leds-pca955x: Using pca9552 16-bit LED driver at slave address 0x60
[    4.500000] leds-pca955x 3-0060: gpios 238...253
[    4.500000] leds-pca955x 11-0060: leds-pca955x: Using pca9552 16-bit LED driver at slave address 0x60
[    4.510000] leds-pca955x 11-0060: gpios 222...237
[    4.520000] usbcore: registered new interface driver usbhid
[    4.520000] usbhid: USB HID core driver
[    4.550000] dps310 3-0076: iio:device1: sensor 'dps310'
[    4.560000] NET: Registered protocol family 10
[    4.560000] Segment Routing with IPv6
[    4.560000] NET: Registered protocol family 17
[    4.570000] 8021q: 802.1Q VLAN Support v1.8
[    4.580000] input: gpio-keys-polled as /devices/platform/gpio-keys-polled/input/input1
[    4.610000] ubi0: default fastmap pool size: 25
[    4.620000] ubi0: default fastmap WL pool size: 12
[    4.620000] ubi0: attaching mtd3
[    4.630000] ubi0: scanning is finished
[    4.650000] ubi0: attached mtd3 (name "obmc-ubi", size 31 MiB)
[    4.650000] ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    4.670000] ubi0: min./max. I/O unit sizes: 1/256, sub-page size 1
[    4.670000] ubi0: VID header offset: 64 (aligned 64), data offset: 128
[    4.670000] ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
[    4.670000] ubi0: user volume: 4, internal volumes: 1, max. volumes count: 128
[    4.700000] ubi0: max/mean erase counter: 4/0, WL threshold: 4096, image sequence number: 1358235831
[    4.700000] ubi0: available PEBs: 106, total reserved PEBs: 398, PEBs reserved for bad PEB handling: 0
[    4.700000] ubi0: background thread "ubi_bgt0d" started, PID 506
[    4.700000] ubi4: default fastmap pool size: 25
[    4.700000] ubi4: default fastmap WL pool size: 12
[    4.730000] ubi4: attaching mtd7
[    4.770000] ubi4: scanning is finished
[    4.810000] ubi4: attached mtd7 (name "alt-obmc-ubi", size 31 MiB)
[    4.810000] ubi4: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
[    4.810000] ubi4: min./max. I/O unit sizes: 1/256, sub-page size 1
[    4.810000] ubi4: VID header offset: 64 (aligned 64), data offset: 128
[    4.810000] ubi4: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
[    4.810000] ubi4: user volume: 3, internal volumes: 1, max. volumes count: 128
[    4.860000] ubi4: max/mean erase counter: 2/1, WL threshold: 4096, image sequence number: 1358235831
[    4.860000] ubi4: available PEBs: 111, total reserved PEBs: 393, PEBs reserved for bad PEB handling: 0
[    4.860000] ubi4: background thread "ubi_bgt4d" started, PID 511
[   14.800000] block ubiblock4_0: created from ubi4:0(rofs-c88ee715)
[   14.800000] console [netcon0] enabled
[   14.800000] netconsole: network logging started
[   14.800000] input: gpio-keys as /devices/platform/gpio-keys/input/input2
[   14.800000] rtc-rv8803 11-0032: setting system clock to 2017-09-28 17:29:24 UTC (1506619764)
[   14.800000] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
[   14.800000] devtmpfs: mounted
[   14.800000] Freeing unused kernel memory: 284K
[   14.800000] This architecture does not have kernel memory protection.
[   17.690000] systemd[1]: Failed to insert module 'autofs4': No such file or directory
[... 119 lines cut ...]
[ 7316.500000] [<802427dc>] (device_release_driver) from [<80240400>] (unbind_store+0x84/0x104)
[ 7316.500000] [<80240400>] (unbind_store) from [<8023fc24>] (drv_attr_store+0x30/0x3c)
[ 7316.500000] [<8023fc24>] (drv_attr_store) from [<8014541c>] (sysfs_kf_write+0x50/0x54)
[ 7316.500000] [<8014541c>] (sysfs_kf_write) from [<80144b14>] (kernfs_fop_write+0xe4/0x1e4)
[ 7316.500000] [<80144b14>] (kernfs_fop_write) from [<800e24b4>] (__vfs_write+0x3c/0x128)
[ 7316.500000] [<800e24b4>] (__vfs_write) from [<800e2730>] (vfs_write+0xb0/0x178)
[ 7316.500000] [<800e2730>] (vfs_write) from [<800e28e4>] (SyS_write+0x4c/0xa0)
[ 7316.500000] [<800e28e4>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 7316.500000] Code: e591c004 e1510008 e2855001 05918000 (e580c004)
[ 7316.500000] ---[ end trace ddac546d95d88e49 ]---
[ 7327.910000] ir35221 4-0071: Failed to read PMBUS_MFR_ID
[ 7327.910000] ir35221: probe of 4-0071 failed with error -71
[ 7328.120000] ir35221 5-0071: Failed to read PMBUS_MFR_ID
[ 7328.120000] ir35221: probe of 5-0071 failed with error -71

NEW BMC LEVEL
=============

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r9-0-gebef540"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r9"```

jk-ozlabs, 2017-09-28:
@cbostic: could you try a kernel with extra debugging config enabled? This may help to track down the source of memory issues. I'd suggest:
```
DEBUG_OBJECTS
DEBUG_OBJECTS_FREE
DEBUG_SLAB
DEBUG_SLAB_LEAK
```

cbostic, 2017-09-29:
Enabled memory debug.  The fail recreates easily on w51 but I see that the fsi_unscan takes a different path prior to the bad access:  In this situation the data shows that the hub master structure's 'upstream' field is unexpectedly kfree'd.  A subsequent deref of hub->upstream causes the prob

```[  227.590000] hub_master_remove >> hub@925e5800 idx:2 h->up@925e5c00
[  227.690000] ### fsi_master_unregister >>  master@925e5800 m->up@925e5c00
[  227.690000] ### fsi_master_unregister 1  master@925e5800 m->up@925e5c00
[  227.690000] ### hub_master_release freeing hub@925e5800 idx:-1
[  227.790000] ### hub_master_release << hub@925e5800 after free
[  227.790000] ### fsi_master_unregister <<  master@925e5800 m->up@6b6b6b6b
[  227.790000] hub_master_remove 1 hub@925e5800 h->up@6b6b6b6b
[  227.900000] hub_master_remove 1a hub@925e5800 h->up@6b6b6b6b
[  228.010000] hub_master_remove 1b hub@925e5800 h->up@6b6b6b6b
[  228.010000] hub_master_remove 1c hub@925e5800 h->up@6b6b6b6b
[  228.130000] Unable to handle kernel paging request at virtual address 6b6b6cdf
[  228.130000] pgd = 92498000
[  228.130000] [6b6b6cdf] *pgd=00000000
[  228.130000] Internal error: Oops: 5 [#1] ARM
[  228.130000] CPU: 0 PID: 1973 Comm: sh Not tainted 4.10.17-7b73717adf28e56beb926bc47c589dfe7daa5038 #1
[  228.130000] Hardware name: ASpeed SoC
[  228.130000] task: 925630a0 task.stack: 92518000
[  228.130000] PC is at hub_master_remove+0x110/0x18c
[  228.130000] LR is at hub_master_remove+0xfc/0x18c
[  228.130000] pc : [<802d14b0>]    lr : [<802d149c>]    psr: 60000013
[  228.130000] sp : 92519ba8  ip : 00000007  fp : 92519bc4
[  228.130000] r10: 925e5c54  r9 : 00000000  r8 : 00000034
[  228.130000] r7 : 0ccccb60  r6 : 925e5800  r5 : 00000000  r4 : 8052daa8
[  228.130000] r3 : 6b6b6b6b  r2 : 6b6b6b6b  r1 : 6b6b6b6b  r0 : 00000000
[  228.130000] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  228.130000] Control: 00c5387d  Table: 92498008  DAC: 00000055
[  228.130000] Process sh (pid: 1973, stack limit = 0x92518188)
[  228.130000] Stack: (0x92519ba8 to 0x9251a000)
[  228.130000] 9ba0:                   925e5c00 925e5c34 00000000 80603868 92519bec 92519bc8
[  228.130000] 9bc0: 80245598 802d13ac 9d91b4cc 925e5c00 806034f8 9c5fa800 80646cec 00000000
[  228.130000] 9be0: 92519bfc 92519bf0 8024564c 80245460 92519c1c 92519c00 80243868 80245638
[  228.130000] 9c00: 925e5c54 925e5c00 925e5c08 9c5fa800 92519c5c 92519c20 80242598 8024378c
[  228.130000] 9c20: 801dfe60 800c1bac 0000003c 00000002 9d1bfcc0 925e5c00 00000000 802d0160
[  228.130000] 9c40: 0ccccb60 00000034 00000000 94e82454 92519c74 92519c60 802426dc 802423b0
[  228.130000] 9c60: 00000000 00000000 92519c84 92519c78 802d0178 802426c4 92519cac 92519c88
[  228.130000] 9c80: 802407fc 802d016c 9d1bfcc0 9250b850 0ccccb60 9c5fa800 00000000 802d0128
[  228.130000] 9ca0: 92519cc4 92519cb0 802d014c 802407a4 00000000 00000000 92519cec 92519cc8
[  228.130000] 9cc0: 802407fc 802d0134 9d30a8c0 9d1bfcd0 92519d04 94e82000 00000000 94e82000
[  228.130000] 9ce0: 92519d04 92519cf0 802d12cc 802407a4 8052daa8 00000000 92519d24 92519d08
[  228.130000] 9d00: 802d13f4 802d127c 94e82400 94e82434 00000000 80603868 92519d4c 92519d28
[  228.130000] 9d20: 80245598 802d13ac 9d91b4cc 94e82400 806034f8 9c482600 80646cec 00000000
[  228.130000] 9d40: 92519d5c 92519d50 8024564c 80245460 92519d7c 92519d60 80243868 80245638
[  228.130000] 9d60: 94e82454 94e82400 94e82408 9c482600 92519dbc 92519d80 80242598 8024378c
[  228.130000] 9d80: 801dfe60 800c1bac 014200ca 00000000 9c9896e0 94e82400 00000000 802d0160
[  228.130000] 9da0: 924dfbc0 00000000 00000000 00000000 92519dd4 92519dc0 802426dc 802423b0
[  228.130000] 9dc0: 00000000 00000000 92519de4 92519dd8 802d0178 802426c4 92519e0c 92519de8
[  228.130000] 9de0: 802407fc 802d016c 9c9896e0 9d30a930 8060dd34 9c482600 00000000 802d0128
[  228.130000] 9e00: 92519e24 92519e10 802d014c 802407a4 00000000 00000000 92519e4c 92519e28
[  228.130000] 9e20: 802407fc 802d0134 9dbe91e0 9c9896f0 9ca9a078 00000002 9dbd4010 92519f78
[  228.130000] 9e40: 92519e64 92519e50 802d0ef8 802407a4 802d0ed0 00000002 92519e7c 92519e68
[  228.130000] 9e60: 8023f180 802d0edc 8023f158 00000002 92519e94 92519e80 801471f4 8023f164
[  228.130000] 9e80: 925e3a80 00000002 92519ed4 92519e98 801468ec 801471b0 00000000 00000000
[  228.130000] 9ea0: 00000817 925e3a8c 92519efc 92461640 80146808 92519f78 00000002 00000000
[  228.130000] 9ec0: 00000002 00000000 92519f44 92519ed8 800e428c 80146814 00000817 80011b94
[  228.130000] 9ee0: 00bb536c 92519fb0 0000270c 000001ff 92519fac 92519f00 80009244 80011ba0
[  228.130000] 9f00: 59ceb101 0be76bc3 59ceb14e 01312d23 59ceb14e 800e44cc 0000221d 00000000
[  228.130000] 9f20: 00000002 92461640 00bb4368 92519f78 00000000 00000002 92519f74 92519f48
[  228.130000] 9f40: 800e4508 800e425c 0000000c 801017e4 92461640 92461640 00000000 00000000
[  228.130000] 9f60: 00bb4368 00000002 92519fa4 92519f78 800e46bc 800e4464 00000000 00000000
[  228.130000] 9f80: 00000002 00bb4368 4dd2e150 00000004 8000a6e4 92518000 00000000 92519fa8
[  228.130000] 9fa0: 8000a540 800e467c 00000002 00bb4368 00000001 00bb4368 00000002 00000000
[  228.130000] 9fc0: 00000002 00bb4368 4dd2e150 00000004 00000002 00000004 00000000 000e912c
[  228.130000] 9fe0: 00000000 7ea397b4 4dc59758 4dcb287c 60000010 00000001 00000000 00000000
[  228.130000] [<802d14b0>] (hub_master_remove) from [<80245598>] (device_release_driver_internal+0x144/0x1d8)
[... cut ...]
[  228.130000] [<801471f4>] (sysfs_kf_write) from [<801468ec>] (kernfs_fop_write+0xe4/0x1e4)
[  228.130000] [<801468ec>] (kernfs_fop_write) from [<800e428c>] (__vfs_write+0x3c/0x128)
[  228.130000] [<800e428c>] (__vfs_write) from [<800e4508>] (vfs_write+0xb0/0x178)
[  228.130000] [<800e4508>] (vfs_write) from [<800e46bc>] (SyS_write+0x4c/0xa0)
[  228.130000] [<800e46bc>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[  228.130000] Code: 1afffffa e5963198 e59621a0 e596119c (e5930174) 
[  228.860000] ---[ end trace 845b9fd1b1a032b6 ]---
```

jk-ozlabs, 2017-10-02:
It's not that the `->upstream` pointer is being free()-ed, it's the whole `struct fsi_master_hub` (setting the pointer, and all other values, to `POISON_FREE` == 0x6b).

It looks like `->release` is being called too early.

jk-ozlabs, 2017-10-02:
We might want to hold a reference on `hub->master.dev` until _remove finishes.

jk-ozlabs, 2017-10-02:
Try this change:

  https://github.com/jk-ozlabs/linux/commit/43c2486501bdd22e246c1bada59ca80fca0d186b

This seems to fix the issue for me.

It may not be the final solution though; we should figure out if the `fsi_master{un,}register()` path in the core should have different semantics with reference counting, as we likely have the same reference counting issue in other master implementations.

cbostic, 2017-10-02:
Thanks Jeremy.   I verified this fixed the issue on my system as well.  Will submit your patch to OpenBMC.

jk-ozlabs, 2017-10-02:
@cbostic: excellent.

After thinking about it, I think this logic makes sense as the final fix. In regards to the API at the moment, the `fsi_master_register()` and fsi_master_unregister()` are symmetrical in that they take and drop a reference to the struct device. Master drivers can add their own reference if necessary.

So, I will add some comments to the core API to explain that, and send a v2 to the openbmc list.

jk-ozlabs, 2017-10-03:
Sent: http://patchwork.ozlabs.org/patch/820715/

closed 2017-11-13 by rfrandse