openbmc/openbmc #2384: SSH server - risk to get stuck due to failure in reverse DNS lookup (suggestion, remove DNS lookup)

guilhermepiccoli, 2017-09-27:
This afternoon I ended up stuck without access to BMC due to a mistake I did:
I added an Austin DNS server to a Pok openBMC, in /etc/resolv.conf.
This fact per se has no issues, but suddenly I lost SSH connectivity there (I wasn't kicked from a SSH session, but couldn't connect anymore).

Reason: SSH server of openBMC seems to use a reverse DNS feature, which was failing due to my misconfiguration of DNS. Since SSH is the only way to access openBMC, this seems fragile enough to me...

Suggestion: change the ssh server config to add "UseDNS no" in order to prevent this issue. I couldn't test it myself since I don't know where the ssh server config lives in openBMC...

When the problem happens, the "ssh -v" log shows we're stuck in:
  debug1: expecting SSH2_MSG_KEX_ECDH_REPLY 

Thanks,

Guilherme

guilhermepiccoli, 2017-09-27:
@anoo1 , @williamspatrick : what do you think about it? I can help testing anything you need.
Thanks for the attention!!

amboar, 2017-10-04:
I've encountered this, I'll send a patch.

guilhermepiccoli, 2017-10-04:
Thanks Andrew =)

guilhermepiccoli, 2017-11-17:
I'm curious, was this fixed anyhow? Thanks!

geissonator, 2017-11-17:
Thoughts @ratagupt or @hramasub ?

amboar, 2017-11-17:
No it's not fixed sorry. I looked at the problem and it's not a simple config change, if i recall we have to patch dropbear to make it work. It's obviously possible, is just more effort than i was expecting

guilhermepiccoli, 2017-11-18:
Oh okay, no problem! It's not a huge bug or something...
Nice to know it's alive though and we're discussing solutions heheh

Thanks!

amboar, 2019-04-02:
I haven't heard of anyone running into this since it's been open, so closing. Re-open if someone hits the problem.

closed 2019-04-02 by amboar