openbmc/openbmc #2387: Add hwmon threshold values

spinler, 2017-09-28:
phosphor-hwmon provides the ability to specify threshold values via its config file, which when populated will show up in D-Bus and emit signals when crossed.  We need to add these threshold values for basically everything with decent values.

rfrandse, 2017-10-14:
<https://gerrit.openbmc-project.xyz/7327> Add OCC default threshold values
Resolves  #2387 Add hwmon threshold values

mine260309, 2017-10-23:
@spinler 
Foxconn @ryandcyu tried to make the similar config for G2 and found it's not working.
The root cause is that `phosphor-hwmon` actually does not read the `label` for threshold values, instead, it just match the `tempX_input` in hwmon.

For example, if we have label `107` for `temp1`:
```
# cat temp1_*
42000
107
```
And we have below config:
```
MODE_temp1 = "label"
LABEL_temp107 = "core1"
WARNLO_temp107 = "10000"
WARNHI_temp107 = "80000"
```
The WARNLO/HI are not working at all.
```
    "/xyz/openbmc_project/sensors/temperature/core1": {
      "Scale": -3,
      "Unit": "xyz.openbmc_project.Sensor.Value.Unit.DegreesC",
      "Value": 42000
    },
```

But if the config is changed to:
```
WARNLO_temp1 = "10000"
WARNHI_temp1 = "80000"
```
it works:
```
    "/xyz/openbmc_project/sensors/temperature/core1": {
      "Scale": -3,
      "Unit": "xyz.openbmc_project.Sensor.Value.Unit.DegreesC",
      "Value": 42000,
      "WarningAlarmHigh": 0,
      "WarningAlarmLow": 0,
      "WarningHigh": 85000,
      "WarningLow": 10000
    },
```

So we need to fix `phosphor-hwmon` to make threshold work for label as well.

chiabinglee, 2017-10-23:
Hi Lei (@mine260309), i did the sensors threshold value test recently, found the same issue like you mentioned.

Suggest that threshold value checking change to (>= or <=),
`(*realIface.*Thresholds<T>::alarmLo)(value <= lo);`
`(*realIface.*Thresholds<T>::alarmHi)(value >= hi);`

For example, FAN Tech sensor sets warning low threshold value 0, it must warning low when FAN is not stuff or zero speed, the fan speed could not less than 0.

Please refer to the IPMI Spec "35.13.3 High-going versus Low-going Threshold Events" for more detail.
[IPMI Specification, V2.0, Rev. 1.1: Document
](https://www.intel.com.tw/content/www/tw/zh/servers/ipmi/ipmi-second-gen-interface-spec-v2-rev1-1.html)

mine260309, 2017-10-23:
Thanks @chiabinglee it's really helpful information.

Btw, do you have a fix of `phosphor-hwmon` and if yes could you kindly help to submit your patch?

spinler, 2017-10-23:
@mine260309 Please see the commits that end at https://gerrit.openbmc-project.xyz/#/c/7314/.

closed 2017-10-24 by geissonator