openbmc/openbmc #2388: Witherspoon: BMC is unresponsive

yadlapati, 2017-09-28:
BMC is not responding to any comments. Host is still up and running. I can not get any response even on BMC debug console.

BMC level:
#cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r9-0-gebef540"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r9"

[ 1478.680000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 1482.210000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 1493.190000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 1493.550000] occ-hwmon occ2-dev0: failed to get OCC poll response: -19
[ 1597.250000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 1600.290000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[ 1612.310000] occ-hwmon occ1-dev0: failed to get OCC poll response: -19
[ 1612.840000] occ-hwmon occ2-dev0: failed to get OCC poll response: -19
[ 1744.070000] ------------[ cut here ]------------
[ 1744.070000] kernel BUG at /home/bmcbuild/releases/obmc910/Builds/ob910.1739.20170927o/output/build/tmp/work-shared/witherspoon/kernel-source/kernel/cgroup.c:918!
[ 1744.070000] Internal error: Oops - BUG: 0 [#1] ARM
[ 1744.070000] CPU: 0 PID: 1 Comm: systemd Not tainted 4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1
[ 1744.070000] Hardware name: ASpeed SoC
[ 1744.070000] task: 9e421b60 task.stack: 9e4a0000
[ 1744.070000] PC is at cgroup_migrate_prepare_dst+0x1c8/0x3c4
[ 1744.070000] LR is at 0x9de7e000
[ 1744.070000] pc : [<8006e200>]    lr : [<9de7e000>]    psr: 80000093
[ 1744.070000] sp : 9e4a1d90  ip : 9dd36300  fp : 9e4a1ddc
[ 1744.070000] r10: 9dcb86c0  r9 : 9dd1325c  r8 : 9dcb86dc
[ 1744.070000] r7 : 9e4a1da0  r6 : 9e4a1de0  r5 : 93e21c00  r4 : 9dd13240
[ 1744.070000] r3 : 9de50bd0  r2 : 93fa0124  r1 : 942e818f  r0 : 8d3008e5
[ 1744.070000] Flags: Nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1744.070000] Control: 00c5387d  Table: 9de5c008  DAC: 00000055
[ 1744.070000] Process systemd (pid: 1, stack limit = 0x9e4a0188)
[ 1744.070000] Stack: (0x9e4a1d90 to 0x9e4a2000)
[ 1744.070000] 1d80:                                     00000005 8062dc04 9dd13268 9e4a1db8
[ 1744.070000] 1da0: 9e4a1da0 9e4a1da0 9e4a1ddc 9e4a1db8 8006ddd0 90551c20 90551c20 00000001
[ 1744.070000] 1dc0: 93e21c00 9e4a1de0 00000005 93e21c00 9e4a1e0c 9e4a1de0 8006fc10 8006e044
[ 1744.070000] 1de0: 9dd13268 9dd13268 00000000 90551c20 93ecea80 8062de28 00000001 00000005
[ 1744.070000] 1e00: 9e4a1e54 9e4a1e10 8006ff98 8006fb88 9edfdc80 800c9c88 9ddac700 90551c28
[ 1744.070000] 1e20: 800c9c88 00000959 9de44160 00000000 00000000 9e4a1f78 90485400 00000000
[ 1744.070000] 1e40: 00000000 00000000 9e4a1e64 9e4a1e58 80070074 8006fdac 9e4a1e94 9e4a1e68
[ 1744.070000] 1e60: 8006cf60 80070064 00000000 00000000 00000005 00000005 9e4a1f78 93ecea80
[ 1744.070000] 1e80: 00000005 9e4a1f78 9e4a1ed4 9e4a1e98 80144b14 8006cf2c 00000000 00000000
[ 1744.070000] 1ea0: 0000081f 93ecea8c 9e4a1efc 93d6be60 80144a30 9e4a1f78 00000005 00000000
[ 1744.070000] 1ec0: 00000005 00000000 9e4a1f44 9e4a1ed8 800e24b4 80144a3c 0000081f 80011b94
[ 1744.070000] 1ee0: 562959c0 9e4a1fb0 00000005 562c3458 9e4a1fac 9e4a1f00 80009244 80011ba0
[ 1744.070000] 1f00: 59cd0776 1d34cfae 59cd0776 1d34cfae 59cd0776 800e26f4 00000407 00000000
[ 1744.070000] 1f20: 00000005 93d6be60 562959c0 9e4a1f78 00000000 00000005 9e4a1f74 9e4a1f48
[ 1744.070000] 1f40: 800e2730 800e2484 00000011 800ffa0c 93d6be60 93d6be60 00000000 00000000
[ 1744.070000] 1f60: 562959c0 00000005 9e4a1fa4 9e4a1f78 800e28e4 800e268c 00000000 00000000
[ 1744.070000] 1f80: 00000005 562959c0 562c3458 00000004 8000a6e4 9e4a0000 00000000 9e4a1fa8
[ 1744.070000] 1fa0: 8000a540 800e28a4 00000005 562959c0 0000006a 562959c0 00000005 00000000
[ 1744.070000] 1fc0: 00000005 562959c0 562c3458 00000004 00000005 00000004 0000006a 7e93d850
[ 1744.070000] 1fe0: 00000000 7e93d5c0 00000000 76bdc8a4 80000010 0000006a e22ef1db 6b45edfe
[ 1744.070000] [<8006e200>] (cgroup_migrate_prepare_dst) from [<8006fc10>] (cgroup_attach_task+0x94/0x14c)
[ 1744.070000] [<8006fc10>] (cgroup_attach_task) from [<8006ff98>] (__cgroup_procs_write.constprop.21+0x1f8/0x2b8)
[ 1744.070000] [<8006ff98>] (__cgroup_procs_write.constprop.21) from [<80070074>] (cgroup_procs_write+0x1c/0x20)
[ 1744.070000] [<80070074>] (cgroup_procs_write) from [<8006cf60>] (cgroup_file_write+0x40/0xec)
[ 1744.070000] [<8006cf60>] (cgroup_file_write) from [<80144b14>] (kernfs_fop_write+0xe4/0x1e4)
[ 1744.070000] [<80144b14>] (kernfs_fop_write) from [<800e24b4>] (__vfs_write+0x3c/0x128)
[ 1744.070000] [<800e24b4>] (__vfs_write) from [<800e2730>] (vfs_write+0xb0/0x178)
[ 1744.070000] [<800e2730>] (vfs_write) from [<800e28e4>] (SyS_write+0x4c/0xa0)
[ 1744.070000] [<800e28e4>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 1744.070000] Code: e5863004 e24bd028 e89daff0 e7f001f2 (e7f001f2) 
[ 1744.070000] ---[ end trace eb141cf792431ef8 ]---
[ 1744.890000] Internal error: Oops - undefined instruction: 0 [#2] ARM
[ 1744.890000] CPU: 0 PID: 987 Comm: dbus-daemon Tainted: G      D         4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1
[ 1744.890000] Hardware name: ASpeed SoC
[... 36 lines cut ...]
[ 1744.890000] df60: 00000001 00000000 7ed28810 0000004c 40000000 00000000 9dfcdfac 7ed28884
[ 1744.890000] df80: 7ed288e4 00000048 9dfcdfa4 9dfcdf98 80315a68 803159e4 00000000 9dfcdfa8
[ 1744.890000] dfa0: 8000a540 80315a5c 7ed28884 7ed288e4 00000048 7ed28884 40000000 00000000
[ 1744.890000] dfc0: 7ed28884 7ed288e4 00000048 00000129 0000004c 007a6264 00000120 7ed288c4
[ 1744.890000] dfe0: 48dde994 7ed287f8 48dc37d4 48b518c0 60000010 00000048 1dd10e24 94ba1201
[ 1744.890000] [<80040b7c>] (__wake_up_common) from [<80040ecc>] (__wake_up_sync_key+0x3c/0x48)
[ 1744.890000] [<80040ecc>] (__wake_up_sync_key) from [<803af75c>] (unix_write_space+0x5c/0x84)
[ 1744.890000] [<803af75c>] (unix_write_space) from [<803179cc>] (sock_wfree+0x80/0x88)
[ 1744.890000] [<803179cc>] (sock_wfree) from [<803ad45c>] (unix_destruct_scm+0x74/0x7c)
[ 1744.890000] [<803ad45c>] (unix_destruct_scm) from [<8031adb8>] (skb_release_head_state+0x60/0x8c)
[ 1744.890000] [<8031adb8>] (skb_release_head_state) from [<8031c020>] (skb_release_all+0x1c/0x34)
[ 1744.890000] [<8031c020>] (skb_release_all) from [<8031c118>] (consume_skb+0x34/0x9c)
[ 1744.890000] [<8031c118>] (consume_skb) from [<803ae450>] (unix_stream_read_generic+0x6fc/0x7a4)
[ 1744.890000] [<803ae450>] (unix_stream_read_generic) from [<803ae5d4>] (unix_stream_recvmsg+0x48/0x54)
[ 1744.890000] [<803ae5d4>] (unix_stream_recvmsg) from [<80313130>] (sock_recvmsg+0x28/0x2c)
[ 1744.890000] [<80313130>] (sock_recvmsg) from [<80313d0c>] (___sys_recvmsg+0x98/0x124)
[ 1744.890000] [<80313d0c>] (___sys_recvmsg) from [<80315a24>] (__sys_recvmsg+0x4c/0x78)
[ 1744.890000] [<80315a24>] (__sys_recvmsg) from [<80315a68>] (SyS_recvmsg+0x18/0x1c)
[ 1744.890000] [<80315a68>] (SyS_recvmsg) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 1744.890000] Code: 00000000 00000000 00000000 00000000 (ffffffe0) 
[ 1744.890000] ---[ end trace eb141cf792431ef9 ]---
[ 1776.100000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[ 1776.110000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
[ 1779.260000] ftgmac100 1e660000.ethernet eth0: no vlan ids left to set
[ 1779.260000] ftgmac100 1e660000.ethernet eth0: NCSI interface up

geissonator, 2017-09-28:
@amboar - not seeing Sam on here to assign to, please help looking into this one.

mdmillerii, 2017-09-28:
```
/* Hierarchies should be linked in the same order. */
BUG_ON(cgrp1->root != cgrp2->root);
```

The illegal instruction is because we are branching to `hrtimer_work` which is a `struct work` not a function from a wakeup along the unix socket read path in `dbus-daemon`.

amboar, 2017-10-04:
@sammj ping

sammj, 2017-10-04:
Hi. I'll have a look into this, doesn't necessarily look NCSI related though.
Does this reproduce?

geissonator, 2017-10-04:
I think this was related to a memory corruption issue in Eddies patch in 10-0-r9.  We updated in 10-0-r11 and haven't seen it since.  https://gerrit.openbmc-project.xyz/#/c/7056/ was the patch sets we were picking up, latest is in 10-0-r11

sammj, 2017-10-04:
> I think this was related to a memory corruption issue in Eddies patch in 10-0-r9.

Is this the patch above or something else?
Sounds like it's worth trying to recreate on 10-0-r11 then.

geissonator, 2017-10-05:
Yeah, we haven't seen this with his latest changes, closing this one out.

closed 2017-10-05 by geissonator