openbmc/openbmc #2389: No API to clear GUARD, leading to bricked systems

cdeadmin, 2017-09-28:


geissonator, 2017-09-29:
# **STATUS**
9/29 - Discussed in team meeting, overall a fairly simple change but looking for the right place to put the new object.  Will finalize on Monday.

anoo1, 2017-10-02:
Per team discussion, we'll create a /org/openpower/control/gard object path under the pnor updater and implement FactoryReset on it.
On the documentation, describe the assertions and assumption about gard, like the one where gard is a preserved partition.

mtritz, 2017-10-05:
- [ ] Create a new object path under the host updater at /org/openpower/control/gard
- [ ] Implement the FactoryReset interface under the new path
- [ ] Create a reset() function - when FactoryReset is called, clear the gard partition

rfrandse, 2017-10-16:
<https://gerrit.openbmc-project.xyz/7352> Host updater: implement GUARD factory reset
Resolves  #2389 No API to clear GUARD, leading to bricked systems

stewart-ibm, 2017-11-08:
Please reopen this until there's some documentation on the API. I only seem to find the implementation.

zahrens, 2017-11-15:
Opened #2637 to track documentation

closed 2017-11-16 by zahrens